Jyers/Marlin #1792: Z-offset issue: Nozzle hit and scrape the bed

venuraherath, 2022-01-17:
Hi,

I was using E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin. Even after setting the correct Z-offset and double-checking with manually leveling, the nozzle started scraping the bed ([it scraped a good portion of my bed](https://www.reddit.com/r/ender3v2/comments/s5vwer/print_mark_cant_be_removed_with_ipa_or_dishsoap/?utm_source=share&utm_medium=web2x&context=3) :'( ). 
Then I went back to E3V2-ManualMesh-5x5-v4.2.2-v2.0.1.bin version (after disconnecting bltouch) and things were perfectly fine. I feel it has to do something with the E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin. I was using the original Creality Bltouch. Please look into the matter. If you need any additional info please let me know.
Thank you so much for making the firmware! 
Best regards,
Venura

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-01-17:
@venuraherath 
Sorry but not a firmware issue, I am using UBL version from #1694.
You must resume your Z-Offset calibration procedures. This must always be done at the printing temperature, especially for the brass nozzle which at 210° expands from 0.04mm to 0.07mm at 260°C (checked and measured!)...
In ManualMesh the result is misleading, you don't have the same precision for the mesh and you can be slightly higher.

venuraherath, 2022-01-17:
@tititopher68-dev  Thank you for the reply. I was running it at "prepare PLA default setting" with both heated nozzle and bed. I will try again.

venuraherath, 2022-01-17:
> @venuraherath Sorry but not a firmware issue, I am using UBL version from #1694. You must resume your Z-Offset calibration procedures. This must always be done at the printing temperature, [...]

@tititopher68-dev is there a way to increase preheat temp setting of the precompiled version? Thank you again!

tititopher68-dev, 2022-01-17:
@venuraherath 
Control Menu -> Temperature -> Warmup, PLA, ABS, etc...

venuraherath, 2022-01-17:
> @venuraherath Control Menu -> Temperature -> Warmup, PLA, ABS, etc...

Thank you so much!

Sonicboom247, 2022-01-20:
@tititopher68-dev 
should the new Discussions be added to new issues as an option ?
Do you want to ask a question? Are you looking for support? Please don't post here. Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

mitomenai, 2022-02-09:
This mod has many, many issues. 

1. Disable stepper doesn't work.
2. Change filament causes Z axis motor to ignore stop switch and plunge the print head with full force into the bed, potentially shattering the glass bed and causing major damage to the motor.
3. Default printing behavior was altered without warning: Print head moves along the top for unknown reasons. It hit a bunch of clips that hold the glass.

Once the print head mashed by clips, I went ahead and reverted my firmware back. Way too much unexpected and dangerous behavior.

closed 2022-04-02 by venuraherath