Jyers/Marlin #1795: Leveling with CR Touch (Ender3v2), mesh vs manual level with probe not consitent

AlexHowser, 2022-01-17:
### Description

While the Values in the created mesh are e.g. for  the lower left corner 0.08, the value shown in manual leveling with probe is 0.034 low (i assume -0.034, see attached images.)
The Problem is that it makes it kind of difficult to believe in either one of these values. Am I missing something here?

### Steps to Reproduce

1.  Level Bed
2. Create Mesh
3. View Mesh
4. Check on the red corners in Manual leveling menue (use probe)

**Expected behavior:** [What you expect to happen]
Values are consistent

**Actual behavior:** [What actually happens]
Values arent constent
[image]
[image]

tititopher68-dev, 2022-01-17:
@AlexHowser 
The probe does not measure the same point when you create a Mesh and when you level the 4 corners of the Bed. Note that depending on the quality of the springs, this can also sometimes vary from one print to another...

AlexHowser, 2022-01-18:
@tititopher68-dev  thanks, i did not know that, could have saved me a couple of hours :D
i have changed my springs allready.
 What i did is:

-set all bedsprings to a simmilar level
- Preheat bed to 70 and nozzle to 240
- set z offset with paper and microsteps
- save z offset
- do a manual paperlevel with the manual leveling menu
- create new mesh
- whole mesh turns up red

now i amrunning out of options..

turner850, 2022-01-21:
Those numbers look great.. That bed looks almost perfectly level. I don't think it's possible to get anything better than that on an Ender 3..

.01-.08 is pretty much in tolerance for an ender 3 from my experience.

closed 2022-01-25 by AlexHowser