Jyers/Marlin #1809: Thermal Runaway not working ? 2.0.1

damiank388, 2022-01-21:
### Description

E3V2-BLTouch-3x3-v4.2.2-v2.0.1.bin - Powered my printer through USB port to test Thermal Runaway. Printer won't shut off. 
I've tested two other 10s Pro v2 the same way, and  thermal runaway was detected, error msg shown. 

### Steps to Reproduce

1. Connect phone charger through micro usb port  
2. Set nozzle temp to 200
3. Wait for thermal runaway error

**Expected behavior:**  
Thermal runaway detected- both my 10s pros did that (1 is stock firmware one is on TM3D)
https://www.youtube.com/watch?v=z5s2iXBpnrc&t=290s

**Actual behavior:** Nothing happens

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-01-21:
@damiank388 
Not a firmware but a Creality Hardware management issue.  The Ender3V2 are poorly designed in terms of grounding.  They fixed this defect on the new Ender 2 Pro and 3 S1.  Personally, I redid a correct grounding and this type of Thermal Runaway Test works.

damiank388, 2022-01-21:
So is this type of test flawed? on V2 or will thermal runaway protection work if heating cartridge was out of heating block getting red hot, printer powered normal way?

tititopher68-dev, 2022-01-21:
@damiank388 
Youtubers (TH3D) have done tests of this type and the answer is negative...

marcribe, 2022-01-21:
> @damiank388 Not a firmware but a Creality Hardware management issue. The Ender3V2 are poorly designed in terms of grounding. They fixed this defect on the new Ender 2 Pro and 3 S1. Personally, I redid a correct grounding and this type of Thermal Runaway Test works.

What did you do to correct the grounding?

tititopher68-dev, 2022-01-21:
@marcribe 
https://github.com/Jyers/Marlin/issues/1709#issuecomment-1001594679

marcribe, 2022-01-26:
@tititopher68-dev Thank you very much!

saskford, 2022-02-28:
I just made a new post here about the same problem. I was running v1.3.5 but found that it failed a thermal runaway test. I then loaded some old firmware I had from another site and the thermal test immediately triggered a message. I tried another older version from this site and once again the test did not trigger a warning. I don't see how this can be a Creality issue.

tititopher68-dev, 2022-02-28:
@saskford 
Thermal runaway failed in test isn't a firmware issue. Since Marlin v2.0.6.x, Marlin changed the process to be more accurate. This issue is a hardware issue. I modified grounding my printer following the example of the Ender 2 Pro.  And since my modifications, the printer passes the test successfully.