Jyers/Marlin #1811: Support for Creality Ender 3v2 with Board v4.3.1

crbn60, 2022-01-21:
### Description

Please add compiles for the Reality `v4.3.1` control board. I received a new Ender 3v2, purchased from a Canadian reseller this week, and that is the version of the control board inside it.

#### Additional Information

[image]

tititopher68-dev, 2022-01-21:
@crbn60 
Use the sources #1694 
Replace platformio.ini with platformio.ini in the Ender3V2-MB v431.zip file.
Make your choice of your configuration's files in the Ender3V2-MB v431.zip file.

[Ender3V2 -MB v431.zip](https://github.com/Jyers/Marlin/files/7917130/Ender3V2.-MB.v431.zip)

And compile your own firmware -> 

Update Python3:
Installation of Python : https://www.python.org/downloads/release/python-3101/
While you're in python setup windows set the checkbox add Python 3.10 to PATH. Enable all optional features. Go to Advanced, check add python to environment variables, set install directory `C:\Program Files\Python310`

Installation VSC : 

1) Install VSCODE: `https://code.visualstudio.com/download`
2) Install 2 extensions: `PlatformIO IDE` and `Auto Build Marlin`. To do this, once VSCODE is installed, Menu file-> Preferences -> Extension and look for one of the extensions. Validate when found. In the main area, the extension appears: Click on install. Do the same for the other.
3) Go to #1694 download .zip file (source files) or : https://github.com/Jyers/Marlin/files/7808535/Marlin-bugfix-JyersUI-v202h.zip
4) Unzip the source files and the Ender3V2-MB v431.zip file
5) Replace platformio.ini with platformio.ini in the Ender3V2-MB v431 folder.
6) Go to the folder: Configuration Files in the Ender3V2-MB v431 folder and choose the folder corresponding to your configuration. 
7) Copy the 2 files `Configuration.h` and `Configuration_adv.h`  in the Marlin folder.
8) Open VSCODE.
9) File Menu -> Open Folder and choose THE folder that includes your Marlin, Configuration Files, etc...
10) At the bottom of the VSCODE information bar, click the Checkmark next to the House icon. The compilation starts.

11) The compiled .bin file is located at .pio/build/STM32.../

Dank84, 2022-01-22:
Followed what you said and end up getting a compiling error. Error: Unknown environment names 'STM32F103RCT6_creality'

Dank84, 2022-01-22:
I flashed with E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1 and am printing fine with a CRTouch

tititopher68-dev, 2022-01-22:
@Dank84 
You will have problems in your impressions! When using v422 firmware, the X and Y axes are reversed, i.e. the X axis (v422) corresponds to the Y axis (v431)....
The CPU STM32F103RCT6 that equips your motherboard is limited to 256K instead of 512K on a V427 or V422 (STM32F103RET6) !
You can recompile your firmware with the Platformio.ini from the sources in #1694 (so without replacing it). The CPU is managed in the same way but with less Flash Memory!
I would be you, I would ask Creality for a v427 motherboard on the pretext of a problem with yours, because having flash memory divided by 2 is not cool and you will not be able to benefit from new features to come.

Creality does anything in the management of their Motherboard. There are v431s with an STM32F103RET6 and other v431s with an STM32F103RCT6! and no information provided. !

crbn60, 2022-01-23:
Is there a drop-in replacement board from a third-party for this unit that would make things easier instead of having to try and get Creality to send me an older board?

crbn60, 2022-01-23:
> printing fine

I would think the above would not be the case if the X & Y axis were reversed on @Dank84's board…

Dank84, 2022-01-23:
nothing seems to be reversed. Its printing wonderfully.

I ordered a 4.2.7 board off amazon @crbn60

crbn60, 2022-01-23:
> nothing seems to be reversed. Its printing wonderfully.
> 
> I ordered a 4.2.7 board off amazon @crbn60

Why did you order a replacement if it's working fine as is?

@tititopher68-dev how do we know how our specific board is setup?

Dank84, 2022-01-23:
for the memory as per tititopher

tititopher68-dev, 2022-01-23:
@crbn60 
Simply because motherboard v431 is the motherboard of Ender6 and under Marlin, we have the pin's file...after perhabs just Creality has inverted the steps motors cables...

Dank84, 2022-01-23:
@tititopher68-dev do you know the pins I need to switch if that is the case. The 4.2.7 board is arriving today

Dank84, 2022-01-23:
just received the 4.2.7 board and it has a STM32F103RCT6

crbn60, 2022-01-23:
> Simply because motherboard v431 is the motherboard of Ender6 and under Marlin, we have the pin's file...after perhabs just

I need to print a case for a Pi Zero, then I'll take the cover off and document what is connected to where on the board.

tititopher68-dev, 2022-01-23:
@Dank84 @crbn60 
Either way, a lot of things will have to be changed.  According to feedback received today, using with an STM32F103RCT6 a firmware compiled for an STM32F103RET6 will present a danger.
When flashing the motherboard or in the start-up phase or with Octoprint, the heating of the BEd or the Hotend can be triggered and cause an uncontrolled Thermal Runaway.  Fires are reported...

Currently, the latest v422 and v427 motherboards come with this CPU with 256K instead of 512K :(!

It is therefore imperative to modify the `stm32f1.ini` file in the `ini` folder by adding just above:

```
#
# BigTree SKR Mini E3 V2.0 & DIP / SKR CR6 (STM32F103RET6 ARM Cortex-M3)
#
```
this:

```
#
# Creality (STM32F103RCT6)
#
[env:STM32F103RCT6_creality]
extends                     = STM32F103RET6_creality
board                       = genericSTM32F103RC
```

Dank84, 2022-01-23:
thanks @tititopher68-dev 

im getting Error: Please specify platform for 'STM32F103RCT6_creality' environment

tititopher68-dev, 2022-01-24:
@Dank84 
Why add `platform = ${common_stm32.platform}` ???

Dank84, 2022-01-24:
without that line in the stm32f1.ini , when I try to compile it returns

Error: Please specify platform for 'STM32F103RCT6_creality' environment

tititopher68-dev, 2022-01-24:
@Dank84 
Vous avez bien seulement ajouter ces lignes pour avoir ceci : 
```
#
# Creality (STM32F103RET6)
#
[env:STM32F103RET6_creality]
extends                     = stm32_variant
board                       = genericSTM32F103RE
board_build.variant         = MARLIN_F103Rx
board_build.offset          = 0x7000
board_upload.offset_address = 0x08007000
build_flags                 = ${stm32_variant.build_flags}
                              -DMCU_STM32F103RE -DHAL_SD_MODULE_ENABLED
                              -DSS_TIMER=4 -DTIMER_SERVO=TIM5
                              -DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8
build_unflags               = ${stm32_variant.build_unflags}
                              -DUSBCON -DUSBD_USE_CDC
extra_scripts               = ${stm32_variant.extra_scripts}
                              pre:buildroot/share/PlatformIO/scripts/random-bin.py
monitor_speed               = 115200
debug_tool                  = jlink
upload_protocol             = jlink
#
# Creality (STM32F103RCT6)
#
[env:STM32F103RCT6_creality]
extends                     = STM32F103RET6_creality
board                       = genericSTM32F103RC

#
# BigTree SKR Mini E3 V2.0 & DIP / SKR CR6 (STM32F103RET6 ARM Cortex-M3)
#
```

tititopher68-dev, 2022-01-24:
@Dank84 
You must modify too :
Marlin/buildroot/share/PlatformIO/ldscripts/creality.ld
with : 
```
MEMORY
{
  ram (rwx) : ORIGIN = 0x20000000, LENGTH = 64K - 40
  rom (rx)  : ORIGIN = 0x08007000, LENGTH = 256K - 28K
}
```

Dank84, 2022-01-24:
@tititopher68-dev 
copying everything above still returns the error

Error: Please specify platform for 'STM32F103RCT6_creality' environment

tititopher68-dev, 2022-01-24:
@Dank84   
You are using v2.0.2h sources, because it does not work with v2.0.1

Dank84, 2022-01-24:
@tititopher68-dev 
Yes using what you suggeseted above, I removed and redownloaded all files to confirm, same error

Sonicboom247, 2022-01-24:
You can leave the environment empty and from the Platformio Project Tasks, scroll down and select your board and then select build, just to be certain.
 [image]

tititopher68-dev, 2022-01-24:
@Dank84 
I do a commit now here : https://github.com/Jyers/Marlin/issues/1811#issuecomment-1019610296

Dank84, 2022-01-24:
@tititopher68-dev it still returns the same error

tititopher68-dev, 2022-01-24:
@crbn60 @Dank84 
OK all fixed in v2.0.2i #1694

crbn60, 2022-01-25:
> @crbn60 @Dank84
> OK all fixed in v2.0.2i #1694

That's fantastic! I'll try and build it later tonight. Wish me luck…

Dank84, 2022-01-25:
@tititopher68-dev thanks, just flashed, went to home all axis and it seems to be ignoring the Y axis end stop. Runs the printer into it till it kills the auto home.

tititopher68-dev, 2022-01-25:
@Dank84 
This is where Creality is driving everyone crazy right now.  The V431 is normally the Ender6 card but with another CPU!  So I don't understand Creality.  They have 4.2.3 which are improved 4.2.2 with an STM32F103RET6 and identical pin coding...
In your case, it's difficult to do without having the card, especially since Creality does not communicate on it.
So declare a v422 (V4) card in configuration.h, you compile and then mount the Z axis. Position the Hotend in the middle in X and Y and only ask for a Home X, then a Home Y...You can enjoy it  to test the respective Endstops by manually triggering them.  Cheer up!

Dank84, 2022-01-25:
@tititopher68-dev  Just tested with the manual control and it seems the x and y are reversed x=y y=x This is when i declare in config.h 4.3.1

I will try what you suggested above. 

If you want the board I can send it to you, like I said I ordered a 4.2.7 and it came with the RCT6

tititopher68-dev, 2022-01-25:
@Dank84 
That's what I imagined, a DIY or a cable or driver inversion.  So declare #define ...V4 (so default config) and it should work.

Dank84, 2022-01-25:
@tititopher68-dev You are right, declaring V4 and can now complete a Home All. 

Thanks Again, Like I said if you want the board I will send it to you .

crbn60, 2022-01-28:
Did you manage to build firmware for the v4.3.1 board @Dank84 (and maybe you could share?) Thanks!

tititopher68-dev, 2022-01-28:
@crbn60 
You must use the source #1694 for RCT6, and you don't modify the type of MB. You keep V4 (for v422, not V431 !!). Because isn't not really a v431 MB from Ender6 but like a v422 with a RCT6 CPU.

Dank84, 2022-01-28:
@crbn60 ya everything is working. Hit me up on discord and I will help you out. Check your email for my info

crbn60, 2022-01-28:
Thanks to @tititopher68-dev for figuring it out and @Dank84 for walking me through building the firmware, I'm happily printing away.

Should this issue stay open until the patch is in master?

gary9872, 2023-01-04:
I just compiled it.. Does it work perfectly or are still some issues? 

[image]