Jyers/Marlin #1828: [Feature Request] Change Filament Load/Unload Speed via UI

nuker1110, 2022-01-26:
One issue I'm running into since changing to smaller nozzles than 0.4mm is that of the extruder grinding at certain speeds due to backpressure from the smaller opening. This is particularly glaring when using the automated Filament Load and Unload features.

Adding a "Maximum Filament Change Speed" setting next to the Load/Unload Lengths would be awesome.

tititopher68-dev, 2022-01-26:
@nuker1110 
For the next build (after v2.0.2i #1694), I will integrate a unload_feedrate, a fast_load_feerate and a slow_load_feedrate in a new menu Filament (after build v2.0.2i #1694).

jhigueras, 2022-03-13:
Really interested in this feature! Sometimes the initial extrusion (when unloading) is a bit fast... and the extruder jumps.