Jyers/Marlin #1837: Remaining print time shows 0:00 with only 3/4 of the print finished

dogunter, 2022-02-06:
I understand printing time is estimated from a line-at-a-time reading of the gcode file but I had a weird thing happen just now. With only 3/4 of a small print finished, the remaining time displayed 0:00. The piece went on to finish printing over the next 10 minutes with no other issues, all the while displaying the 0:00 remaining time value.

### Steps to Reproduce

1. Print a part, no supports, brim for bed adhesion, 20% infill (triangles)
2. Notice remaining time drop to 0:00 long before piece finishes printing.

**Expected behavior:** Remaining time to be a non-zero value.

**Actual behavior:**Remaining time is 0:00 long before part finishes printing

Currently using the E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v2.0.1 firmware.

tititopher68-dev, 2022-02-06:
@dogunter 
You should be more specific.  Printing from Octoprint or from the printer.  Also check the Gcodes file, it is from this that the estimate is made.  A reading error and the estimate will no longer be made with a time automatically reduced to 0. If the printing is done from Octoprint, a communication error, even temporary, causes the same thing.
I advise you to compile your own firmware with the sources of v2.0.2i in #1694.  The v2.0.1 contains the bugs of the 1.3.5 because built partly on this version instead of the v1.3.5b (why? Mystery)

dogunter, 2022-02-06:
Sorry about that, yes, I'm printing from Octoprint. I wasn't aware that it would make a difference where the gcode file was coming from.

I've briefly looked at the instructions for compiling my own. It's daunting but I may give it a try. Thanks.

dogunter, 2022-02-27:
This was an OctoPrint issue, not a jyers issue. Closing.

closed 2022-02-27 by dogunter