Jyers/Marlin #1838: Clicking issue

kshaub103, 2022-02-06:
I have had my printer working after replacing just about every component. Have been have good success but now I'm running into the clicking issue sounds is random but i believe something is not right. The print is also shifting layers. Here's a video https://user-images.githubusercontent.com/96362402/152702510-b1e312db-c783-4c3b-a7a1-e0b4df0241eb.MOV
[image]

kshaub103, 2022-02-06:
https://user-images.githubusercontent.com/96362402/152702940-414dbd59-b14b-40b0-b9d7-d8a5e59a80ed.MOV

Cal-LF, 2022-02-06:
@kshaub103 The clunking and layer shift was a failed main board on mine.

kshaub103, 2022-02-06:
Interesting since i just replaced the board, what would be cause this not sure it's worth a new board every time I want to print. How can we trace the root cause?

Cal-LF, 2022-02-06:
What board is installed? If v427, then try increasing the Vref on the driver that appears to be slipping.

kshaub103, 2022-02-06:
Yes, it's the newer board V4.2.7, by chance can you explain or lead me into changing those settings?

tititopher68-dev, 2022-02-06:
@kshaub103 
The clicking sound you hear is a step skip at the extruder motor.  If you're using PTFE tubing going all the way to the nozzle, it's pretty standard after a while.  On your video, it is characteristic, we see the tube go up with each retraction.  You have a plug that is created in the Hotend.  The problem and the fix are in a CHEP video (about 1 year ago).  You must put a ring at the level of the cooling radiator of the Hotend and separate the PTFE tube into 2 parts or use a magnetic coupler from Hotends.fr.
 Regarding the layers shift, obviously we can think of an overheating of the drivers.  But the TMC2225 are known to heat up less and above all to avoid layers shift in Z. But many other things can be the cause...such as the Marlin base of version v2.0.1 which can cause timer and  PWM at motor level.

Cal-LF, 2022-02-06:
https://reprapworld.com/customer/service/technical_information/adjusting_stepper_driver_current_on_the_ender3_v2/
There are videos on youtube that will help also.
Check for a blockage in the filament path as @tititopher68-dev  stated.

kshaub103, 2022-02-06:
Hotends.fr. That is very interesting, I can refresh my French that I had for 8 years of schooling. Lol I looked at the metal cir clip and it wasn't holding the ptfe tubing from the filament runout sensor (btw that sensor always errors not sure why) another issue.

Cal-LF, 2022-02-06:
@kshaub103 I use the motion sensor with no issues. The settings for motion sensor is set in the config.h file and the compiled.

kshaub103, 2022-02-06:
> The problem and the fix are in a CHEP video (about 1 year ago). 

By chance do you know the title of the video cant seam to find it.

kshaub103, 2022-02-06:
> such as the Marlin base of version v2.0.1 which can cause timer and PWM at motor level.

Is there an update?
As always everyone thanks for the input. I'm a live long learner and enjoy my printer.

kshaub103, 2022-02-06:
> @kshaub103 I use the motion sensor with no issues. The settings for motion sensor is set in the config.h file and the compiled.

Nice, I will note that, is the a guide? Thanks for that tip.

kshaub103, 2022-02-06:
Time to get the multi meter out. Any suggestions with the raspberry pi power affecting the board? Do I need to cover the 4pin on the usb? Or is there a better option? I can't seam to get the tape to cover the contact.

tititopher68-dev, 2022-02-07:
@kshaub103 
> Do I need to cover the 4pin on the usb?
Yes, no other option.

> Is there an update?
> As always everyone thanks for the input. I'm a live long learner and enjoy my printer.
Yes,  look at #1694

> The clicking sound you hear is a step skip at the extruder motor. If you're using PTFE tubing going all the way to the nozzle, it's pretty standard after a while. On your video, it is [...]
Video -> https://youtu.be/7tCxO17XZtw

kshaub103, 2022-02-07:
> @kshaub103 
> 
> > Do I need to cover the 4pin on the usb?
> 
> Yes, no other option.
> 
>sounds good 
> 
> > Is there an update?
> 
> Yes,  look at #1694
> 
> thanks I would be lost without the guidance
> i will be hopeful up and running soonish 
> 
>

kshaub103, 2022-02-11:
Still working on the printer here's what I have now I adjusted the stepper drivers to X,Y,Z: 1.22V
E: 1.39V now it won't print. https://user-images.githubusercontent.com/96362402/153612617-eee7e7f6-55eb-4bb0-826b-74d6a9a1f581.MOV

kshaub103, 2022-02-11:
![Uploading 153612677...]

kshaub103, 2022-02-11:
Didn't hear stepper skipping but I run a test before the video and it was clicking like before. I have adjusted the ptfe tube as well

kshaub103, 2022-02-11:
Great so I need a new board again

tititopher68-dev, 2022-02-11:
@kshaub103 
And now, I think your step motor are out !

kshaub103, 2022-02-11:
Maybe time to scrap this printer ?

tititopher68-dev, 2022-02-11:
@kshaub103 
> Great so I need a new board again
Not necessary I think ! But change all your step motors yes. I see another windows with exactly the same issue on a Ender5 but only on Y axis. It's was the step motor !
You can check your step motor. Déconnecte tous les step motors and just reconnecte X axis and test it with a move command. Do the same for all axis. If you ear a clicking sound during test, the step motor is defect

kshaub103, 2022-02-11:
https://user-images.githubusercontent.com/96362402/153619002-996bebad-fac4-4e2c-8304-a255c2ec2aea.MOV

kshaub103, 2022-02-11:
More then confused I'm not sure what to do.

tititopher68-dev, 2022-02-11:
@kshaub103 
With your new video (more explicit that the one), I advise you start by checking your step motors.
You disconnect all the cables from your motor steps.  Then you reconnect only one cable (for example the X-Axis).  You place your hotend against the Endstop (only for the X-Axis, for the Y it will be your Bed, etc...).  You make an X move to the end and in the opposite direction.  If you hear a click, the step motor is faulty (probably a ball in the internal bearing).  You repeat the procedure for all axes.

kshaub103, 2022-02-11:
Ok, this makes sense to me. I believe it's in the X Axis. But I will check them all. As for the main board should I replace or Hope I didn't damage it by adjusting the stepper drivers

Cal-LF, 2022-02-12:
@kshaub103 If the mainboard drivers failed, the steppers will not work. If all are still working, reduce the vref for each to less than 1.1 and see how it goes.

kshaub103, 2022-02-12:
Ya I don't believe I killed the main board. I ordered new stepper motor hoping it the X. Tho I ordered one of each. I haven't isolated the motor but it the last failed attempt the X stop as all other motors continued to work. Plus in my video it seam to happen more when the X is moving then any other

kshaub103, 2022-02-13:
Any idea what's going on with my bl touch https://user-images.githubusercontent.com/96362402/153733115-5587885c-1f81-41b3-9967-bdad2059cb75.MOV

kshaub103, 2022-02-13:
https://user-images.githubusercontent.com/96362402/153733134-b6ae9c6a-e8e5-4bc7-8c74-7bcc5d7952b4.MOV

tititopher68-dev, 2022-02-13:
@kshaub103 
There is an allen head grub screw at the top for adjusting the magnetic force between the pin and the solenoid. you could try adjusting that. it seems like your pin is falling out of the magnetic field under fast movements and needs a bit more of the magnetic coil exposed in order to hold the pin up.

kshaub103, 2022-02-13:
https://user-images.githubusercontent.com/96362402/153734329-665df977-f8da-4992-b10d-11532f7fba54.MOV

kshaub103, 2022-02-13:
Printer ends print after bl touch probs. Got one print with no errors now I can't get it to work again?

kshaub103, 2022-02-13:
https://user-images.githubusercontent.com/96362402/153738268-4329cfea-706d-4a47-a936-7afe760026c8.MOV

kshaub103, 2022-02-13:
What is the issue now? I feel like I'm throwing money down the drain I have two main board, stepper motors, end switch's. Things will work for a print then bam another issue.

kshaub103, 2022-02-13:
Well I turned in off and removed one screw on the X stepper motor holding the mount for my web camera for octoprint and it stopped I'm not sure what the screw in the stepper motor would cause that well I probably will order another motor but let's see if I can get a print off. I am working on update jyers but would like to get the printer working

kshaub103, 2022-02-13:
How to you set offset for bl touch? I'm configuring my on firmware

kshaub103, 2022-02-13:
Got the new firmware installed but I'm receiving a homing error when I try to home printer

closed 2022-02-13 by kshaub103