Jyers/Marlin #1850: Live Adjustment Z-Offset not moving Z axis

e-SonOfAnder, 2022-02-13:
### Description

When attempting to use the Live Adjustment for setting the Z Offset after installing JyersUI 2.0.1 on an Ender 3 V2 with a CR Touch installed, adjustments to the Z Offset value are ignored, and the stepper motor is never activated to move the print head. This is an initial setup of the firmware, and I have done some searching around, but could find no other reports of this issue as described (outside of people attempting to compile their own versions of Marlin).

### Steps to Reproduce

1. Flash JyersUI 2.0.1 per instructions
2. Go to Prepare -> Z-Offset
3. Select Home Z Axis to move the print head to Z = 0
3. Select Live Adjustment to enable it (if it is not already enabled)
4. Attempt to move the Z Offset either up or down

**Expected behavior:**
Print head moves up or down based on the value of the Z Offset as it is being adjusted

**Actual behavior:**
Nothing. Even if you commit the value by clicking the rotary knob, the value does not apply. I have a rough idea of what my Z Offset was with the stock firmware prior to switching to JyersUI, but I would prefer not to risk a bed crash if I can avoid it should the value be slightly different after moving the printer around a couple of times.

**Additional Information**
I am using the pre-compiled binary file E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin from the Latest Release.

Hardware is an Ender 3 v2 with CR Touch installed and connected to the proper dedicated port on the motherboard. Only other modification to the printer is upgraded bed springs, otherwise it is a stock printer. With the stock firmware, the Z Offset adjustment did work as expected. The auto mesh leveling appeared to work correctly, and I am in the final stage before attempting my first print with the new firmware.

Cal-LF, 2022-02-13:
Try the latest release #1694.
DL the precompiled 202h. [E3V2-Pre-compiled-firmwares-v2.0.2h.zip](https://github.com/Jyers/Marlin/files/7808537/E3V2-Pre-compiled-firmwares-v2.0.2h.zip)

e-SonOfAnder, 2022-02-13:
Partial success with 2.0.2h. With Live Adjust enabled, after making an adjustment to the Z-Offset value and then pushing the rotary encoder to confirm, it then moves the nozzle. I would have expected it to move the nozzle actively while changing the Z-Offset value. I do not have a dev environment set up to compile the firmware myself, but if I get some time later to do that I will see about compiling 2.0.2i.

e-SonOfAnder, 2022-02-13:
New bug, after setting the Z-Offset successfully using the above method, I went to do a test print, and the print completely ignored the Z-Offset, and attempted to print where the CR Touch left the nozzle. Additionally, even though I have G29 A G29 J in my startup gcode, the printer did not do the four point mesh tilt, it tested once at the center, and once at the X/Y = 0 point, and then started to attempt the print.

tititopher68-dev, 2022-02-13:
@e-SonOfAnder 
It is better to use version 2.0.2i while waiting for v2.1.0a (for Thursday I think).
There is no live movement for the Z-Offset at this time.  Unless you are using Micro Step Up/Down.
I have no issues with applying the Z-Offset during the print.
Finally your startGcode seems to need to be reviewed.
In UBL, after the G28, you must have a:
 G29 L0;  to load the mesh stored on slot 0
 G29A;  to activate UBL mode
 G29 J2;  for autotilt 4 points / only J -> autotilt 3 points!

MicroSur, 2022-02-14:
@e-SonOfAnder 
My personal opinion: UBL is a strange thing ) You have to re-mesh after setting new Z-offset.
ABL is easier and enough.

drdansyco, 2022-02-19:
e-SonOfAnder. I am having the same issue. I had followed Phil's YouTube video to a t. The z axis does not move.

tititopher68-dev, 2022-02-19:
@drdansyco 
> e-SonOfAnder. I am having the same issue. I had followed Phil's YouTube video to a t. The z axis does not move. I tried the 1.3.4. it doesn't work either.

Read above, not a firmware issue!  Just a normal behavior.  The Z-Offset adjustment is done very well, but differently.  You are confusing live movement and live adjust!  Live movement allows you to move the axes when you turn the rotary knob.  Live Adjust allows you to move the Z axis after clicking on the rotary button.  This is to be used as an approach phase.  The fine adjustment is then carried out by acting on MicroStep Up/Down (and there, the operation is similar to live movement).
Unlike other firmwares using the Babystep method, here the step is 0.01mm instead of 0.025mm.

drdansyco, 2022-02-19:
> @drdansyco
>
>
> Read above, not a firmware issue! Just a normal behavior. The Z-Offset adjustment is done very well, but differently. You are confusing live movement and live adjust! Live movement [...]

Well i sat and hit the micro step and still nothing. I didnt use the live adjustment. after the micro step of about -.58 and started the print the z offset was not there

algonao, 2022-03-14:
It may be that when you apply the negative z offset, the endstop for the z axis is the activated.

e-SonOfAnder, 2022-03-14:
> @drdansyco
>
>
> Read above, not a firmware issue! Just a normal behavior. The Z-Offset adjustment is done very well, but differently. You are confusing live movement and live adjust! Live movement [...]

I disagree, this is a firmware issue. The bug as reported is still active and valid. If you have to make the adjustment and then confirm it with pressing the rotary knob, that's not a live adjustment; that's just an adjustment. The expected behavior from something called "Live Adjustment" is that the adjustments are made... live. The microstep would be to fine tune the adjustment once you've made the large adjustment from the primary method.