Jyers/Marlin #1854: Can't build on MOTHERBOARD BOARD_CREALITY_V427 using various default and suggested environments.

ixendrax, 2022-02-17:
### Description

Can't build on MOTHERBOARD BOARD_CREALITY_V427 using various environments. 

### Steps to Reproduce

Assign configuration with `#define MOTHERBOARD BOARD_CREALITY_V427`and otherwise same config as UBL-BLTouch-10X10 Configuration_adv.h and Configuration.h

1. Assign configuration with `#define MOTHERBOARD BOARD_CREALITY_V427`and otherwise same config as UBL-BLTouch-10X10 Configuration_adv.h and Configuration.h
2. Attempt build on default environment

**Expected behavior:** Build successful. 

**Actual behavior:** Error:
```Error: Build environment 'STM32F103RET6_creality_maple' is incompatible with BOARD_CREALITY_V427. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple```

I've then tried all of the suggested STM32F103RE environments (As I have a 512K Board) and received similar errors about missing FAN_SOFT_PWM:
```In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL.cpp:30:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2019:6: error: #error "FAN_SOFT_PWM is required for your board. Enable it to continue."
 2019 |     #error "FAN_SOFT_PWM is required for your board. Enable it to continue."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2019:6: error: #error "FAN_SOFT_PWM is required for your board. Enable it to continue."
 2019 |     #error "FAN_SOFT_PWM is required for your board. Enable it to continue."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2019:6: error: #error "FAN_SOFT_PWM is required for your board. Enable it to continue."
 2019 |     #error "FAN_SOFT_PWM is required for your board. Enable it to continue."
      |      ^~~~~
*** [.pio/build/STM32F103RE_creality/src/src/HAL/STM32/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RE_creality/src/src/HAL/STM32/HAL_SPI.cpp.o] Error 1
*** [.pio/build/STM32F103RE_creality/src/src/HAL/STM32/MarlinSerial.cpp.o] Error 1
```

#### Additional Information
From what I can find online I suspect this has to do with a Marlin build issue and creating new STM32 environments but really I'm asking what else I can do?

Is there any harm in setting FAN_SOFT_PWM for the 4.2.7 boards?

tititopher68-dev, 2022-02-18:
@ixendrax 
Just in platformio.ini, rename `STM32F103RET6_creality_maple` by S`TM32F103RE_creality_maple` or by `STM32F103RE_creality`
I advise you to use `STM32F103RE_creality`.
Concerning `FAN_SOFT_PWN`, modify in Configuration.h : `//#define FAN_SOFT_PWM` by `#define FAN_SOFT_PWM`

ixendrax, 2022-02-18:
Sure no issue. Just wasn't sure if it was ok to define FAN_SOFT_PWM. 

All works now thanks tititopher68-dev.

closed 2022-02-18 by ixendrax

Ali229, 2022-03-27:
Hey guys,
Is there a reason this isn't pushed into Jyers bugfix brand?