Jyers/Marlin #1860: Cannot compile after getting latest

Ali229, 2022-02-20:
### Description

Receiving error while compiling

### Steps to Reproduce

1. Get latest
2. Use bltouch 10x10 configs
3. Compile

**Expected behavior:**
Build should compile

**Actual behavior:**
Receive error:
_Error: Build environment 'STM32F103RET6_creality_maple' is incompatible with BOARD_CREALITY_V4. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple_

#### Additional Information
I believe this is due to
BOARD_CREALITY_V4 line added to Configuration.h for bltouch 10x10

Cal-LF, 2022-02-20:
@Ali229 The maple environment is not used with Jyers.

tititopher68-dev, 2022-02-20:
@Ali229 
Just look at #1854

zoomming, 2022-02-23:
@Ali229 Maybe select a MOTHERBOARD, e.g. V4.2.7? By default Configuration.h is targeting V4...

KiLLie2k, 2022-03-04:
> 

your amazing! thanks much!!!!!