Jyers/Marlin #1872: All menu Icons missing

toothyc, 2022-02-26:
### Description

After flashing latest firmware with either my custom configuration or the predefined examples leaves the printer without any icons

### Steps to Reproduce

1. compile firmware

**Expected behavior:** Menu icons visible 

**Actual behavior:** No icons

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-02-26:
@toothyc 
Please edit `dwin_set.h` in `src\lcd\e3v2\common`
and modify `//#define USE_STOCK_DWIN_SET` by `#define USE_STOCK_DWIN_SET`
Next, recompile your own firmware or more simply use this sources #1694 (v2.0.2i)...In the v2.0.1, a lot of issue exist.

toothyc, 2022-02-26:
@tititopher68-dev  Thanks for the info i was not aware of the the RC you have created. I will give this ago when my current print job finishes but based on the thread on the discussions page it looks like you have already encountered this before.

Again many thanks for your guidance :)

closed 2022-02-26 by toothyc