Jyers/Marlin #1878: Ender 3 V2 thermal runaway not working

saskford, 2022-02-28:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-02-28:
@saskford 
Thermal runaway failed in test isn't a firmware issue. Since Marlin v2.0.6.x, Marlin changed the process to be more accurate. This issue is a hardware issue. I modified grounding my printer following the example of the Ender 2 Pro.  And since my modifications, the printer passes the test successfully.

saskford, 2022-02-28:
> @saskford Thermal runaway failed in test isn't a firmware issue. Since Marlin v2.0.6.x, Marlin changed the process to be more accurate. This issue is a hardware issue. I modified grounding my [...]

Thanks for the reply.
The only grounding issue I can find online is the power supply not being properly grounded to the frame. I checked this on mine and there is a solid ground between them. Is there somewhere else on the printer that should be grounded better? I really don't feel safe using the printer until the thermal runaway issue is solved.

tititopher68-dev, 2022-03-01:
@saskford 
To be more precise, I resumed (did) the grounding of the motherboard.  This complements that of the chassis.  I should find the pictures here.  This subject has already been discussed.

tititopher68-dev, 2022-03-01:
@saskford
https://github.com/Jyers/Marlin/issues/1709#issuecomment-1001594679

zoomming, 2022-03-02:
https://community.octoprint.org/t/thermal-malfunction-system-stopped-heater-id-e0/41836

saskford, 2022-03-02:
I have checked all the grounding and can find no issues. Does anyone know what version of Jyers firmware would predate Marlin making the thermal protection more accurate. I just want to confirm that is what is causing the problem.
Thanks

saskford, 2022-03-08:
I found this comment on another site where they have a version of Marlin for the Ender 3 V2 similar to Jyers where someone pointed out their thermal runaway protection had stopped working.
"Thank you for your report. Seems that the problem is caused by the new feature THERMAL_PROTECTION_VARIANCE_MONITOR. I will disable it from the configuration."
They did disable it and I can confirm that my thermal protection works fine with this firmware. Since this is a hidden safety issue that can only be discovered if you happen to run the runaway test, which I bet few people do, I believe it should also be turned of in Jyers firmware.

JonSnow88, 2022-03-19:
This is for sure an issue in 2.0.1 hotfix 1, I've been using 1.3.5b for about a year now and specially tested thermal runaway by heating the hotend powering off the printer but keeping the motherboard powered by usb to simulate temp going down while heating.

JonSnow88, 2022-03-19:
> I found this comment on another site where they have a version of Marlin for the Ender 3 V2 similar to Jyers where someone pointed out their thermal runaway protection had stopped working. "Thank [...]

which firmware was this?

saskford, 2022-03-19:
I didn't want to start a firmware battle here by posting a link, but since there is no other way to get it to you here it is.
https://github.com/mriscoc/Ender3V2S1/wiki

tititopher68-dev, 2022-03-19:
@saskford 
No big secret, just comment #define THERMAL_PROTECTION_VARIANCE_MONITOR in configuration_adv.h and after just compile your own firmware.

JonSnow88, 2022-03-19:
> I didn't want to start a firmware battle here by posting a link, but since there is no other way to get it to you here it is. https://github.com/mriscoc/Ender3V2S1/wiki

Thanks, I'm not interested in his closed source firmware. I went back to 1.3.5b (UBL this time) and confirmed thermal runaway is working.

I'll wait until Jyers is finished merging the work provided by @tititopher68-dev.

It could be my method of testing thermal runaway was flawed when on 2.0.1, when i went back to 1.3.5 it also didn't work but to lazy to go back :).

These are the correct steps:

1. Heat Hotend to 200C
2. Wait 1 min to let temps stabilize
3. Plug in usb cable that is plugged into a phone charger
4. Power off printer quickly after step 3  
5. Once temps go below 170C it should trigger thermal runaway.