Jyers/Marlin #1886: Jyers compile issu

Nikos125, 2022-03-06:
Hello, i'm trying to complile jyers firmware and even i have changed nothing i have the following error.
My mainbord is V4.2.2 and chip isSTM32F103RET6. On Platformio i have set  default_envs = STM32F103RET6_creality

> Executing task in folder Marlin-bugfix-JyersUI: C:\Users\nkrit\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RET6_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_RAMPS_14_EFB. Use one of these: mega2560, mega1280
=========================================================================== [FAILED] Took 1.75 seconds =========================================================================== 

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:01.747
====================================================================== 1 failed, 0 succeeded in 00:00:01.747 ======================================================================The terminal process "C:\Users\nkrit\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

tititopher68-dev, 2022-03-06:
@Nikos125 
Why do you set as motherboard in configuration.h BOARD_RAMPS_14_EFB ?????
In configuration.h, you have to define as motherboard  BOARD_CREALITY_V4 ! For a v422 !

Nikos125, 2022-03-06:
@tititopher68-dev Thank you really much. I' m new in compiling. I made the change you told but i had to change the 
default_envs from STM32F103RET6_creality    to    STM32F103RE_creality . I made this and then  the program informed to power on enable FAN_SOFT_PWM.  I made these two changes and the compilation completed thanks to you. 
Is it problem the default_envs change i made?
Thanks again

tititopher68-dev, 2022-03-06:
@tititopher68-dev 
👍

hannibal1923, 2022-03-07:
STM32F103RE_creality or STM32F103RE_creality_maple

hannibal1923, 2022-03-07:
I tried with STM32F103RE_creality. ".bin" file was generated with a warning, but now there no icons on the display. Only text. I am looking into the "DWIN_SET" fix.

tititopher68-dev, 2022-03-07:
@hannibal1923 
Icon -> https://github.com/Jyers/Marlin/discussions/1889#discussioncomment-2304953

Regarding the warning, it is just a reminder of Marlin's Sanity Check so as not to forget to define the drivers TMC2208_STANDALONE (motherboard v422 Ender3V2) or A4988 (for one driver of motherboard v422 Ender3 Max)

hannibal1923, 2022-03-07:
I will try your suggestion. Thanks.