Jyers/Marlin #1895: Screen firmware not flashing

AJinabadMood, 2022-03-10:
I've followed all instructions on the Readme  and the screen doesn't even pretend to flash. I've tried all three files. It just boots up normally with the stock firmware. This morning I found a reddit thread with a handful of other people having the same issue. 
https://www.reddit.com/r/ender3v2/comments/sgdx08/jyers_display_firmware_wont_update_201/?utm_medium=android_app&utm_source=share
Any chance of a fix for this? 
Thanks in advance.

tititopher68-dev, 2022-03-10:
@AJinabadMood 
look at the type of LCD display motherboard you have : 
DWIN motherboard : 
[image]

DACAI motherboard : 
[image]

If you have a DWIN, you must use DWIN_SET, but if you have a DACAI, you must use the folder 'private' in this zip : 
[private.zip](https://github.com/Jyers/Marlin/files/8223737/private.zip)

and following these intructions : 

**Updating the Display**
To update the graphics and icons on the display:

Please check the type of motherboard of your display (you have 2 pictures of the possible motherboard):

Case A : you have an original DWIN Display :

Copy one of the DWIN_SETs from the Firmware Sets folder( Custom pour V203j et sup) to an SD card. Remove the identifier so the folder is just named DWIN_SET. Remove the back cover of the display and insert the card into the MicroSD slot on the back of the display unit's PCB.
Power on the machine and wait for the screen to change from blue to orange (ca. 10 à 20s).
Power off the machine.
Remove the SD card from the back of the display.
Power on to confirm a successful flash.

Troubleshooting :
If flashing is not working (e.g. the blue screen only flashes for a second before the orange screen):
Reformat the SD card in FAT32 with 4096 bytes allocations

Case B : you have a DACAI Display :

Copy the private folder from the Display firmware folder to an SD card. Remove the back cover of the display and insert the card into the MicroSD slot on the back of the display unit's PCB.
Power on the machine and wait. The screen change to blue with a progress bar (picture : Nouvel-ecran-en-cours-de-flash).
Wait Progress bar to 100% and the end of this process. After Power off the machine.
Remove the SD card from the back of the display.
Power on to confirm a successful flash.

Troubleshooting :
If flashing is not working (e.g. the blue screen only wihtout progress bar):
Reformat the SD card in FAT32 with 4096 bytes allocations

AJinabadMood, 2022-03-10:
Houston we have lift off.... sorta. The display flashed but I don't have icons and if I turn the knob I get white squares that don't go away so unless I'm counting my turns I don't know what I'm clicking on.

tititopher68-dev, 2022-03-10:
@AJinabadMood 
Here Houston, here Houston....look at : https://github.com/Jyers/Marlin/discussions/1889#discussioncomment-2304953

karanbskd, 2022-04-17:
I learned that for the DACAI board - You flash LCD first then Firmware. I didn't work another way around(Screen glitch/No text bug).

Maybe I missed it but some tutorials didn't specify this order of operation.

Cristy94, 2022-04-26:
I had the issue, I fixed it in the same way:

* Reformat the SD card in FAT32 with 4096 bytes allocations
* Copy DWIN_SET and the firmware bin (rename it to something else)
* Flash the screen (takes 3 seconds, screen goes blue to orange)
* Flash the mainboard (takes 10 seconds).

Sharkey79, 2022-07-07:
I know (now) this is a Jyers fork and I don't have any experience with Jyers (yet), but I have an original Creality firmware and wanted to update the screen but DWIN_Set wasn't working. Googled it and I got directed here so I thought; "Great! The solution!" without really checking the fork... I know, my mistake... 
So did the screen update. 
See on the screen that it's updating the icons, finished, turn the printer off, take the card out, start the printer....... 
Nothing but black except for the actual temperature of the hot-end and bed. Still no clue, tried it again; same result, and again; nothing. Until accidently I turn the knob. All of a sudden I see pictures. But no text. 
Bummed out I mail Creality cause I can't use my printer this way. They come across a bit reluctant about helping but do it anyway. Turns out, the original Creality firmware doesn't like adjusted pictures. So I installed the official screen firmware from Creality and it's working perfectly again.

dmac282, 2022-08-24:
Hi, what happens if I

> I had the issue, I fixed it in the same way:
> 
> * Reformat the SD card in FAT32 with 4096 bytes allocations
> * Copy DWIN_SET and the firmware bin (rename it to something else)
> * Flash the screen (takes 3 seconds, screen goes blue to orange)
> * Flash the mainboard (takes 10 seconds).
> 
> The SD card was formatted in 32Kb allocation blocks, so after flashing the firmware the screen stayed black.

Hi, I'm stuck on the reality screen, the LCD won't flash, anybody else have this issue? I have the 4.2.2 board

dmac282, 2022-10-30:
@42Willow so if we have the same display board, which I think we do, you have to flash the display first then install marlin onto the board. I got it working awhile back

42Willow, 2022-10-30:
> @42Willow so if we have the same display board, which I think we do, you have to flash the display first then install marlin onto the board. I got it working awhile back

which files did you use to flash the display board? DWIN_SET does nothing for me...

42Willow, 2022-10-31:
@dmac282 sorry for bothering you :/ (last ping a promise) I flashed the display first with the jyers DWIN_SET (I also tried to rename it to something else), then I flashed the latest jyers firmware... Still no icons and white boxes everywhere.

dmac282, 2022-10-31:
@42Willow I used a different firmware for the display, I can't remember what it was called. I'll have to check my computer to see the name of it. It was an update of jyers iirc. I surfed around for a bit and it popped up in some thread

42Willow, 2022-10-31:
after what seems like years of surfing I found the display firmware!
[New.display.firmware.zip](https://github.com/Jyers/Marlin/files/9898190/New.display.firmware.zip)

Blockbuter, 2022-11-21:
> @dmac282 I have the same issue, turns out there are two different LCD boards... When I took the cover off my LCD screen it turns out it is a DACAI board, which means it requires the "private" folder when flashing the screen firmware... See also:
> 
> * [DACAI Screen is the newer model of the DWIN Screen under Firmware Sets.  we need a "private" folder under Display firmware\Firmware Sets #1941](https://github.com/Jyers/Marlin/issues/1941)
> 
> [image]
> 
> ~I haven't been able to find a solution yet but currently I am using the icons here with creality ui marlin firmware V2.1.1 https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples (for example: icons for ender 3 v2 are here https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20V2/LCD%20Files/private) I will update this comment if it works with jyers too (it probably won't)~
> 
> Can't find any icons which work with Jyers and DACAI

You are a lifesaver. I owe you my sanity and my printer. Thank you so much

42Willow, 2023-02-19:
> @42Willow U save my life, after trying hundreds of firmwares, with this file i solve the black screen issue after updating display firmware.
> 
> Hundred of thanks for you. Just a question, where did you find that firmware? is it the default one?

I'm glad it helped :)

I can't remember where I found it, I think it may have been on some reddit post (I didn't save the link). The icons in the "original" folder are the icons of ender 3 v2 stock firmware.

infinitejest, 2023-05-12:
Does this one work with Marlin 2.1. I been trying to use this and it I can see the menus; but non of the icons are showing up. 
Printer : Dacai version Ender 3 V2

42Willow, 2023-05-12:
I'm not sure, my ender 3 is on 2.0.1 Jyers and I'm too scared to update it, now that it's running smoothly. It seems they might have support for DACAI screens now as they added a README explaining it 4 months ago... [This](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20V2) link *may* work.

Mark7888, 2024-03-04:
> after what seems like years of surfing I found the display firmware! [New.display.firmware.zip](https://github.com/Jyers/Marlin/files/9898190/New.display.firmware.zip)

You saved my life! I was looking for this for hours now