Jyers/Marlin #1898: Build custom firmware & configuration of 1.3.5b

antwal, 2022-03-12:
Hi all,

I have to make a modified version of the firmware with the name "E3V2-BLTouch-5x5-v4.2.2-v1.3.5.b.bin"; but I can't find the configuration files used, where can I find them?

i just have to raise the max hotend temperature from 260 to 360 degrees to use a metal hotend, in my case i bought two hotend to test, one is the Creality Spider and the other the Phaetus Dragonfly BMS.

Thanks.

tititopher68-dev, 2022-03-12:
@antwal 
Go to https://github.com/Jyers/Marlin/releases/tag/v1.3.5.b
And download source code (zip).

antwal, 2022-03-12:
> @antwal Go to https://github.com/Jyers/Marlin/releases/tag/v1.3.5.b And download source code (zip).

i have download branch of 1.3.5b, but not found the configuration file of BLTouch 5x5 for 4.2.2

Cal-LF, 2022-03-12:
@antwal You have to modify and compile from the BL Touch 3X3 configs.

closed 2022-07-25 by antwal