Jyers/Marlin #1904: Unable to stop print using LCD Display

cpxazn, 2022-03-16:
### Description

Ender 3 v2 4.2.2

Previously when using jyers firmware version 1.3.4, I was able to stop print with the LCD controls.
Now after flashing with bugfix-JyersUI branch, it is no longer working.

### Steps to Reproduce

1. Pull from bugfix-JyersUI branch
2. Update config changes as necessary
3. Flash firmware
4. Flash LCD display
5. Start print
6. Using LCD controls, stop the print

**Expected behavior:** Print should stop

**Actual behavior:** Print continues

#### Additional Information
[configuration.zip](https://github.com/Jyers/Marlin/files/8271609/configuration.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Cal-LF, 2022-03-16:
Did you reset the firmware?

cpxazn, 2022-03-16:
> Did you reset the firmware?

Are you referring to clearing the EEPROM?

tititopher68-dev, 2022-03-16:
@cpxazn 
Absolutely not a firmware issue, well no it's a firmware issue related only to your configuration files.  Why don't you use the ones provided as a base?
 I've looked at your config files and there's a ton of inconsistency with what you're looking to do.
 Let's take only the case you cite (but you will have many other issues if you don't start from scratch...).  You say print via Octoprint, so this is a so called Host unit and in condmfiguration_adv.h you defined this //#define HOST_ACTION_COMMANDS.  It is certain that by choosing this, you cannot stop your printing with the screen.  If you look at the configuration files provided with the firmware, you will see that this option is active.  However, v2.0.1 still has a lot of issues and I advise you to upgrade to v2.0.2i (#1694) and use the provided configuration files as a base.

cpxazn, 2022-03-17:
@tititopher68-dev
Thanks for the reply, I'll try out v2.0.2i as you have suggested.

closed 2022-03-17 by cpxazn