Jyers/Marlin #1906: Issue with Pause and Change Filament

johncarlson21, 2022-03-17:
### Description

Jyers 2.0.1 BL touch HS 5 x 5
Ender 3v2
Controller flashed
Controller Pause to do a filament change.

I slice and start printing a model, I choose when I want to do a filament change, once i've finished the filament change and the head moves back and starts printing, it seems that the Z is up just a hair and the print starts and it prints, but it seems like the adhesion is no good and you can just break the part at the filament change.

Original ender 3v2 firmware does not move Z up when you click "Pause"

### Steps to Reproduce

1. Slice in Cura
2. Start Print
3. Wait till specific level
4. Press "Pause" on the controller and change the filament
5. Press "Resume"

**Expected behavior:** 
Print should resume just fine and get good adhesion.

**Actual behavior:** 
Print resumes and adhesion is not good, part breaks, and sometimes there are gaps in the outer walls. Especially if you do not check the "Outer before Inner Walls" option in Cura.
Also doesn't seem to matter if I use "Pause" or go into "Tune -> Change Filament"

[image]

tititopher68-dev, 2022-03-17:
@johncarlson21 
it's not a firmware issue ! Why ? Because, Pause and Change filament are managed by Gcodes, so by only code Marlin. I advise you to first compile your own firmware from the sources here (# 1694). You can also have this issue with bad parameters in your slicer (Z-Hope type must be deactivated for example). Personally, using PrusaSlicer, I sometimes have to change filament during printing and I have never had this type of lag.

tititopher68-dev, 2022-03-17:
@johncarlson21 
The Creality firmware does not use the advanced pause features in Marlin and the Marlin codebase of these firmwares is over 2 years old.
I see in your photo that the printer has indeed printed internal fillings in white, for example when you have changed the color (red filament -> white). so I suspect Cura and their sometimes weird Gcodes.

Otherwise no anti backlash installed? Also be careful not to have too high an acceleration in Z. Printing via Octoprint?

I just read that Cura had fixed on version 4.13 problems of height offset during pauses by modifying the generation of their gcodes... Cura was based before on M0 type pauses (as with the Creality firmware) and had problems with M125/M600  Gcodes.

tititopher68-dev, 2022-03-18:
@johncarlson21 
I think you misunderstood what I was talking about with Cura.  Cura produces the file containing the Gcodes needed for printing.  Users were reporting Z offset or print hole issues with the gcode generated by Cura 4.12 on print resume when a pause was included in the gcode, but also when the user himself put  pause printing.  Cura 4.12 (and below) would not always handle pauses correctly with Marlin's Advanced Pause features.  This problem seems to have been fixed with Cura version 4.13.  It is not the printer that is in question, but the sliced ​​file.  Your printer resumes printing at the correct Z-height. We can clearly see the correct printing of infills.  If it was the printer, the infills wouldn't print at the correct Z-height and you'd immediately have spaghetti.  The printer does not choose what it prints.  It stupidly executes the Gcodes of the file one after the other.  It doesn't even load the file.  It directly reads the file step by step from the SD without the memory (which would be physically impossible, given the ridiculous amount of RAM available).  Cura is notorious for not being extremely good at optimizing and consistency of generated Gcodes.  It seems to have improved since 4.13.

johncarlson21, 2022-03-18:
I haven't updated to 4.13 yet as I seen some stuff about a few bugs. I will give this a try, but I also noticed that this is happening on on machine but not another, so it leads me to believe there is something else maybe going on.

sergejbubko, 2022-03-31:
Hi. I also have this or similar issue, but I gave up trying to solve it. This is a Jyerses answer closing my issue in October.

https://github.com/Jyers/Marlin/issues/1347#issuecomment-950262232

Unfortunately I don’t know what he means by ‘set your slicer to layers not height’. I am using PrusaSlicer and never heard about it or be able to google this. Colour change worked for me when I had stock fw maybe even without BL touch. After I updated to Jyers or/and added BL touch or/and started using OctoPrint I can not achieve correct resume print after filament change or pause print using M600 or M125 in PrusaSlicer or using filament change option from printer menu.

@tititopher68-dev do you have any idea, what was Jyers talking about? Thank you!

johncarlson21, 2022-03-31:
So I've since left using Jyers and went to Mriscoc Firmware because I wasn't getting the answers.
I found that the issue is because of a config of minimum Z height. in the Configuration.h file this was the answer:

#define NOZZLE_PARK_Z_RAISE_MIN   0

Jyers has the same config with a value of 5mm

#define NOZZLE_PARK_Z_RAISE_MIN   5   // (mm) Always raise Z by at least this distance

so if you change that to 0 save it and compile your own firmware it will give you the ability to change the z height when using M600

so M600 Z0.00 will actually stay at 0

however if you are using Cura like I was, you actually need to change the plugin script, because if it sees 0, it doesn't put the value in.

tititopher68-dev, 2022-03-31:
@johncarlson21 
I just finished to print this :
[image]
[image]

and no issue !

johncarlson21, 2022-03-31:
I'm guessing that is quite a larger size of text. And that isn't on the side

johncarlson21, 2022-03-31:
Whoops thought I was on a different thread. Yeah it does work. Just not as good for what we are trying to do

tititopher68-dev, 2022-03-31:
@johncarlson21 
But that doesn't change anything! The printer does not decide to print your fillings correctly and then not your walls....It only executes what the Gcode file sends it, gcode by gcode! The print recovery Z height is correct, if there is a problem it is with the slicer as Jyers said, the firmware does not manage the print recovery height...it is the M600 or the M125 and these 2 Gcodes are not linked to a UI, it is the code common to all printers since the marlin 2.0.8.x base and perhaps even before. The basic Creality firmware is based on the Marlin 2.0.1 base and at best the 2.0.6....and maybe start by printing the walls and then the fillings...

johncarlson21, 2022-03-31:
Again, yes this is in the firmware as you see above you will notice what I changed

johncarlson21, 2022-03-31:
I will try to do this a little later tonight.. but for me, my main problem was Retract at Layer Change, which I'm not seeing in your code, maybe a .75mm but mine was after the M600, and then the fact that it lifted the Z height and then went back to the Z height..

johncarlson21, 2022-04-01:
so I had to change just a few things in your file for my temp settings and my bed leveling but I ran yours and I would say mine does not look as bad as yours, however, I notice yours is doing some type of wipe between layers and a bit of retraction right before it resumes the print over the part. Leaving no outer walls.

I use Cura and like I said one of my main issues was the **Retract at Layer Change** which would put a retraction after the M600 and cause it to not have filament in the nozzle. I think yours does retractions before the M600 but I think it is affecting it. My other issue was when it would resume it just wouldn't stick right on the filament change layer, leading me to believe that something was up with the whole Z up and park, then move back and Z down. The adhesion wasn't good.

If you have Cura I can send you a profile I use for Pro PLA and you can test the filament change there. Also what filament are you using? Those are some high temps.

[image]

[image]

ims21, 2022-04-01:
I am using M600 in script in Simplify3D for changing filament on layer without any problem...

johncarlson21, 2022-04-01:
@ims21 yes I have it figured out on my end for the big issue in the slicer but I still wanted the ability to set the Z when running the M600 but you can only set it at 2mm or above. I like it at 0 so I can wipe the nozzle when it moves back.

sergejbubko, 2022-04-02:
> I will try to do this a little later tonight.. but for me, my main problem was Retract at Layer Change, which I'm not seeing in your code, maybe a .75mm but mine was after the M600, and then the fact that it lifted the Z height and then went back to the Z height..

I think that the main issue here is also the lifting. As I mentioned Jyerses comment, I have closed issue, but didn't solve anything. After resume from M600 or M125, the nozzle prints in mid air, which causing this gap.

> If you have Cura I can send you a profile I use for Pro PLA and you can test the filament change there. Also what filament are you using? Those are some high temps.

I am not very familiar with Cura. I've tried to slice the same model in Cura (it was 4.1.0 version) just to prove it is not a slicer issue and have the same results. It was PETG Sunlu.

If I understand correctly, you recommend to build my own version with `#define NOZZLE_PARK_Z_RAISE_MIN 0`. 
Retractions are default values from PrusaSlicer. It happens every time layer changes. Also before the printer resumes from M600 it primes the nozzle. I think the main issue is with height misalignment.

johncarlson21, 2022-04-02:
@sergejbubko yes you would need to build your own firmware changing that, and also if using cura you have to update the post script file for filament change and I'm not sure if putting it in prusa is manually or not.