Jyers/Marlin #1920: Mesh Stuck

thalecko, 2022-03-26:
### Description

When I try to do a new mesh, the printer just stop after 8 points.
I´m tried a lot of versions... For example, E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1

### Steps to Reproduce

1. [First Step]
Level
3. [Second Step]
Create New Mesh

**Expected behavior:** [What you expect to happen]
keep doing the mesh

**Actual behavior:** [What actually happens]
Stop at the point 9.
The axes are stopped and the bltouch is activating and deactivating

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-03-28:
@thalecko 
just look at https://github.com/Jyers/Marlin/issues/1903#issuecomment-1069030581

thalecko, 2022-03-28:
Thanks, first, i will try v2.0.2i (https://github.com/Jyers/Marlin/discussions/1694).

thalecko, 2022-04-07:
I changed the Bltouch and it worked.

closed 2022-04-07 by thalecko