Jyers/Marlin #1925: E3V2-UBL-BLTouch-10x10 moving in x axi touch braket

1achy, 2022-04-02:
I use :

[E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin]

(https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin)

### Description

When i create mesh on the right (x axi) side touch bracket with y support.

 I also set x and y offsets of the probe. 
The distance offet -42 x - 7 y. 
I also changed them but both increasing and decrease x offset the behavior is the same

Also in the leveling menu in the manual tunig if I move point - point I think it does not touch because the reference is the nozzle.

### Steps to Reproduce

1. create new mesh
2. last point on the rigth side touch

1achy, 2022-04-02:
Send M503 for che offset is correct. is it work for leveling?

Send: M503
Recv: echo:; Linear Units:
Recv:   G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E93.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P500.00 R500.00 T1000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00 ; Leveling OFF
Recv: echo:  G29 W I0 J0 Z-0.11550
Recv: echo:  G29 W I1 J0 Z-0.07850
Recv: echo:  G29 W I2 J0 Z-0.06650
Recv: echo:  G29 W I3 J0 Z0.02750
Recv: echo:  G29 W I4 J0 Z0.54500
Recv: echo:  G29 W I0 J1 Z-0.09400
Recv: echo:  G29 W I1 J1 Z-0.06600
Recv: echo:  G29 W I2 J1 Z-0.04550
Recv: echo:  G29 W I3 J1 Z0.02500
Recv: echo:  G29 W I4 J1 Z0.57950
Recv: echo:  G29 W I0 J2 Z-0.05050
Recv: echo:  G29 W I1 J2 Z-0.01800
Recv: echo:  G29 W I2 J2 Z0.00100
Recv: echo:  G29 W I3 J2 Z0.06400
Recv: echo:  G29 W I4 J2 Z0.58450
Recv: echo:  G29 W I0 J3 Z-0.05800
Recv: echo:  G29 W I1 J3 Z0.00850
Recv: echo:  G29 W I2 J3 Z0.03100
Recv: echo:  G29 W I3 J3 Z0.08800
Recv: echo:  G29 W I4 J3 Z0.63100
Recv: echo:  G29 W I0 J4 Z-0.05150
Recv: echo:  G29 W I1 J4 Z0.01600
Recv: echo:  G29 W I2 J4 Z0.03850
Recv: echo:  G29 W I3 J4 Z0.10150
Recv: echo:  G29 W I4 J4 Z0.62950
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H150.00 B45.00 F255
Recv: echo:  M145 S1 H190.00 B55.00 F255
Recv: echo:  M145 S2 H240.00 B70.00 F255
Recv: echo:  M145 S3 H240.00 B70.00 F255
Recv: echo:; Hotend PID:
Recv: echo:  M301 P28.72 I2.62 D78.81
Recv: echo:; Bed PID:
Recv: echo:echo:  M304 P462.10 I85.47 D624.59
Recv: echo:; LCD Brightness:
Recv: echo:  M256 B250
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S1 ; ON
Recv: echo:; Z-Probe Offset:
**Recv: echo:  M851 X-42.00 Y-5.00 Z-3.30 ; (mm)**
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U0.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S0 D0.00 ; Sensor OFF
Recv: ok

1achy, 2022-04-02:
I have dual z axis bracket different 👍 
 i change 
#define X_MAX_POS X_BED_SIZE 15 to 1
#define PROBING_MARGIN 20 to 10

but touch braket

1achy, 2022-04-02:
[image]
[image]

1achy, 2022-04-03:
so do is not correct i change probe offset -42 to -20 the x axi dont hit but now nozzle and probe is not center

1achy, 2022-04-03:
is there different solution ?

Zaaap, 2022-04-18:
So, you have a modified printer that causes the head to hit the modified bracket and think the firmware is bad?

In your case i would modfy the bad bracket to allow more X-Axis travel.

Zaaap, 2022-04-18:
[image]

Zaaap, 2022-04-18:
[image]

SergiuToporjinschi, 2022-04-18:
I would just simple rotate the screw of the last wheel on hot end carriage. Put the nut inside the case an the hexagonal head out side, towards the the back
[image]
[image]