Jyers/Marlin #1940: When using a SD card extender, SD card change is not recognized,

fg0003, 2022-04-23:
I have a Creality Ender 3 v2. When using a SD card extender like this one:
https://www.amazon.de/gp/product/B0855DBVXF
Changing the SD card when the device is switched on, is not recognized.

There is a fix for this error. Modifications in 3 files needs to be done:
https://www.reddit.com/r/ender3v2/comments/lx3vko/enable_simple_swapping_of_sd_cards_when_using_an/

I have successfully made this modification on the latest Jyers/Marlin version 2.0.1. After compiling, the SD card change is working.
I think this should be implemented in the latest release.

tititopher68-dev, 2022-04-23:
@fg0003 
it's a normal behaviour due to your SD card extender! So not a firmware issue !

fg0003, 2022-04-23:
This is just a proposal to improve the firmware and eliminate this error for other users.

Cal-LF, 2022-04-23:
@fg0003 With this firmware you can move off the print selection and back to refresh the print menu.