Jyers/Marlin #1941: DACAI Screen is the newer model of the DWIN Screen under Firmware Sets.  we need a "private" folder under Display firmware\Firmware Sets

marcop300, 2022-04-26:
Ender 3D V2 FW 4.2.2 boards and maybe some newer ones have a newer type of Video Display. you can't just put the folder DWIN_SET on a 8GB SD formatted at 4K  and  have it turn blue/orange. You now need a new folder in the root of a SD card called private. Then when you boot up the Ender with the SD card in the video SD slot. the screen will turn blue and you will see many lines of code that shows that it is loaded the newer ver of FW on the screen.  All these changes do need to added to the Marlin-2.0.1.zip i think. Thanks

tititopher68-dev, 2022-04-26:
@marcop300 
For the time being, Jyers has suspended development of its firmware base. Being the only one able to update his Github. We'll have to wait....even if in the Discussion area, you would have your answer.
In the meantime, I continued development here: https://github.com/tititopher68-dev2/Marlin/tree/new_feature

marcop300, 2022-04-27:
you are Great!! thank you. i will use your code  https://github.com/tititopher68-dev2/Marlin/tree/new_feature/Display%20firmware/DACAI%20Display for FW and Display to see if it fixes my issues. THANK YOU

Snakefrog1, 2022-09-22:
This is the only place I have seen this info. I have this new display and this helped, thank you.

RV-from-be, 2022-11-02:
@42Willow The dev continued the development, currently in version 3.5a Final, available on the Discord JSTECH...Installed on my printer, it's purely fantastic!