Jyers/Marlin #1943: Increasing HEATER_0_MAXTEMP still throws MAXTEMP error

calh, 2022-04-28:
### Description

I have an all metal hot end, and am trying to print Nylon at 300C.  I increased the `HEATER_0_MAXTEMP` to 330 to start with, and then also bumped it up to 350 as a second test.

Both values caused my printer to kill at 297C with the message:

```
Printer Kill Reason:
  Err: MAXTEMP
  Restart Required
```

Under Control -> Temp, the input _does_ actually allow me to set the value past 300, but it seems to consistently fail with this error when it reaches 297.

### Steps to Reproduce

1. Raise HEATER_0_MAXTEMP to 350
2. Set the temp to 300
3. Wait for the error

#### Additional Information

[my_config.zip](https://github.com/Jyers/Marlin/files/8584838/my_config.zip)
[heater_error](https://user-images.githubusercontent.com/1163502/165806007-3c638322-e982-4acb-9029-1c8c01a92628.jpg)

tititopher68-dev, 2022-04-29:
@calh 
If you use the stock heat cartridge and thermistor, it's a normal behaviour. You can't go over 275⁰C without thermal runaway !

calh, 2022-04-29:
@tititopher68-dev, it's currently the stock heater and thermistor, but I just received [this 50W heater and thermistor pack](https://www.amazon.com/dp/B08CVGK8XK), which I'm going to install after my current print is done.

However, it seemed to reach the temps I'm looking for just fine, although it was a little slow to climb past 280C.  It didn't behave like thermal runaway, and grepping through the code where the various temp checks are done, it doesn't make any sense that it should have been triggered.

arxipka79, 2022-12-22:
hello, do you have ender 3 v2?

calh, 2022-12-22:
Yes, it's an ender 3 v2, 4.2.2 mainboard.

arxipka79, 2022-12-23:
can I ask you for this firmware I don't know how to compile firmware

arxipka79, 2022-12-24:
dear friend, can you share this firmware for 300C.I have tried compiling many times and I have errors all the time.I'll be glad if you share.thank you! Отправлено из приложения "Почта" для Windows От: Cal HeldenbrandОтправлено: 22 декабря 2022 г. в 19:32Кому: Jyers/MarlinКопия: arxipka79; CommentТема: Re: [Jyers/Marlin] Increasing HEATER_0_MAXTEMP still throws MAXTEMP error (Issue #1943) Yes, it's an ender 3 v2, 4.2.2 mainboard.

arxipka79, 2022-12-25:
dear friend, can you share this firmware for 300C.I have tried compiling many times and I have errors all the time.I'll be glad if you share.thank you!