Jyers/Marlin #1948: Menu options crash after flash

Sure4Thing, 2022-05-03:
### Description

After flashing, when I select certain menu items like Prepare -> Move or Z-Offset, the screen freezes, the progress bar comes up, and then it crashes back to the main menu

### Steps to Reproduce

1. Flash firmware
2. Click Prepare menu -> Move
3. It crashes

**Expected behavior:** 

The option to move the XYZ axis

**Actual behavior:** [What actually happens]

Screen will freeze and then go back to start

Using:
 - E3V2-BLTouch-5x5-v4.2.2-v2.0.1.bin
 - 8 GB SD Card
 - FAT32
 - 4.2.2 motherboard

Video is too big so here is a link to what is happening
https://drive.google.com/file/d/1bwgJjec9NFsGrZJ__A17E1DwHBJ4EDbb/view?usp=sharing

tititopher68-dev, 2022-05-03:
@Sure4Thing
After every flash, you have to do a Reset to Default, following by a Store Settings ! Golden Rule...
and you can take a look at : https://github.com/tititopher68-dev2/Marlin/tree/new_feature (the last build and  continuation of the very beautiful Jyers Project of which I took over the development )

Sure4Thing, 2022-05-03:
Thanks for your reply.  Im having a gazillion issues getting this beast to accept any firmware and just work.  When I get the screen back on again and back to a stable space Ill give it a shot

closed 2022-05-04 by Sure4Thing