Jyers/Marlin #1950: LCD LED/NEOpixel menu

Nbrooks639, 2022-05-04:
Is it possible to add a menu in LCD to control LEDs or neopixels. is this possible yet? it has been used with other marlin builds currently.

tititopher68-dev, 2022-05-04:
@Nbrooks639 
First it's not an issue but a request for a new functionality, so your post isn't in the correct place ;-)

Please, take a look at : https://github.com/tititopher68-dev2/Marlin/tree/new_feature (the last build and continuation of the very beautiful Jyers Project of which I took over the development )

ANTABY, 2022-05-15:
first sorry for my english.
I am trying to compile the neopixel leds, it is the only thing that I am missing to configure,
I'm using the code new_feature, I undo the line #define NEOPIXEL_LED

and when I compile I get this error:

`Marlin\src\lcd\e3v2\jyersui\dwin.cpp: In static member function 'static void CrealityDWINClass::Value_Control()':
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:6285:31: error: 'leds' was not declared in this scope
 6285 |         if ((valuepointer == &leds.color.r) || (valuepointer == &leds.color.g) || (valuepointer == &leds.color.b))
      |                               ^~~~
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:6286:11: error: 'ApplyLEDColor' was not declared in this scope
 6286 |           ApplyLEDColor();
      |           ^~~~~~~~~~~~~
Compiling .pio\build\STM32F103RE_creality\src\src\libs\buzzer.cpp.o
Compiling .pio\build\STM32F103RE_creality\src\src\libs\crc16.cpp.o
*** [.pio\build\STM32F103RE_creality\src\src\lcd\e3v2\jyersui\dwin.cpp.o] Error 1`

`     #if HAS_COLOR_LEDS
        if ((valuepointer == &leds.color.r) || (valuepointer == &leds.color.g) || (valuepointer == &leds.color.b))
          ApplyLEDColor();
          #if HAS_WHITE_LED
            if (valuepointer == &leds.color.w) ApplyLEDColor();
          #endif
`

Any idea what is happening?

tititopher68-dev2, 2022-05-15:
@ANTABY 
You have just forgot to uncomment in Configuration_adv.h #define  LED_CONTROL_MENU !

ANTABY, 2022-05-15:
But when I uncomment it it doesn't even start compiling.

this is what it gives me,

`                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\firm\marlin-new_feature (1)\marlin-new_feature\marlin\src\feature\leds\neopixel.h:38:10: fatal error: Adafruit_NeoPixel.h: No such file or directory

***************************************************************************
* Looking for Adafruit_NeoPixel.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_NeoPixel.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_NeoPixel.h
*
***************************************************************************

   38 | #include <Adafruit_NeoPixel.h>`

tititopher68-dev2, 2022-05-15:
@ANTABY 
It's not a firmware problem but an issue related to your Platformio installed on VSCode... I just tested and I don't have your issue...

ANTABY, 2022-05-15:
thanks for answering. First of all I forgot to thank you for your excellent work, the truth is that I am very grateful for a piece of firmware, many of the options are very useful, if I cannot make the issue of the LEDs work, it would not be so serious either, that is a pity but do not record.

Regarding installing the library, I installed it before sending these messages and there is no case, I kept having the same problem, that's why I sent the question,
after denying a lot I sent this message,

to guide me use this information

https://www.th3dstudio.com/hc/product-information/ezneo/ezneo-creality-v4-2-x-board-setup/

install the library by hand, install it with VScode, and there is no case I can't make it work. knowing at least that it is not a matter of firmware, I will continue to see where the problem may be, but the truth is that I have run out of ideas :(

[image]

![install](https://i.postimg.cc/VvdfhJD0/Captura-de-pantalla-2022-05-15-171241.jpg
)

tititopher68-dev2, 2022-05-15:
@ANTABY 
I know that since the last Platformio update, some have inexplicable problems and some don't.
 The only thing you can still do is open Platformio.ini.  There is a line starting with `lib_deps = `, add this following: `adafruit/Adafruit NeoPixel@^1.10.4`

ANTABY, 2022-05-15:
> @ANTABY Sé que desde la última actualización de Platformio, algunos tienen problemas inexplicables y otros no. Lo único que aún puede hacer es abrir Platformio.ini. Hay una línea que comienza con `lib_deps = `, agregue lo siguiente:`adafruit/Adafruit NeoPixel@^1.10.4`

https://docs.platformio.org/en/latest/projectconf/section_env_library.html?utm_source=platformio&utm_medium=piohome#lib-deps

Thank you very much for such a prompt response, I was just reading
:P
now i try hope it works
https://docs.platformio.org/en/latest/projectconf/section_env_library.html?utm_source=platformio&utm_medium=piohome#lib-deps

ANTABY, 2022-05-15:
well no, it doesn't want to load the library :(
and it's where it needs to be and everything.

[image]
[image]

tititopher68-dev, 2022-05-15:
@ANTABY 
I think I found. You installed before the latest version while in the Marlin configuration for platformio, the installation was planned but with a different version, 1.8.0...Platformio is looking for this version which has been overwritten by your installation. So in my opinion, you have to force the installation of the 1.8.0 library by doing in a terminal: 
`pio lib install "adafruit/Adafruit NeoPixel@1.8.0"`

ANTABY, 2022-05-16:
@tititopher68-dev2 
So far I haven't been able to get it to work.
I'll uninstall everything tomorrow and do a clean install to see if I can get it to work.
I will be commenting

ANTABY, 2022-05-16:
I haven't had any luck, I've tried everything :O
reinstall vscode, a clean installation of 0 unzipped the .rar and as I unzipped it enabled neopixel, and na, the library is installed, but when I compile

> Processing STM32F103RE_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
In file included from c:\users\egma2\onedrive\desktop\probandoleds\marlin-new_feature\marlin\src\feature\leds\leds.h:39,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../lcd/e3v2/jyersui/dwin_defines.h:31,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:66,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\egma2\onedrive\desktop\probandoleds\marlin-new_feature\marlin\src\feature\leds\neopixel.h:38:10: fatal error: Adafruit_NeoPixel.h: No such file or directory

***************************************************************************
* Looking for Adafruit_NeoPixel.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_NeoPixel.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_NeoPixel.h
*
***************************************************************************

   38 | #include <Adafruit_NeoPixel.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CalledProcessError: Command '"C:\Users\egma2\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" -D__MARLIN_FIRMWARE__ -DNDEBUG -DCPU_TYPE=\"STM32F103RE_creality\" -DHAL_STM32 -DUSBCON -DUSBD_USE_CDC -DTIM_IRQ_PRIO=13 -DADC_RESOLUTION=12 -DMCU_STM32F103RE -DHAL_SD_MODULE_ENABLED -DSS_TIMER=4 -DTIMER_SERVO=TIM5 -DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8 -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\egma2\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 179:
    env.SConscript(item, exports="env")
  File "C:\Users\egma2\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\egma2\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\egma2\OneDrive\Desktop\probandoleds\Marlin-new_feature\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 247:
    apply_features_config()
  File "C:\Users\egma2\OneDrive\Desktop\probandoleds\Marlin-new_feature\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 133:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\egma2\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\egma2\OneDrive\Desktop\probandoleds\Marlin-new_feature\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 216:
    load_marlin_features()
  File "C:\Users\egma2\OneDrive\Desktop\probandoleds\Marlin-new_feature\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 204:
    define_list = run_preprocessor(env)
  File "C:\Users\egma2\OneDrive\Desktop\probandoleds\Marlin-new_feature\buildroot\share\PlatformIO\scripts\preprocessor.py", line 43:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\egma2\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\egma2\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,
================================================================================================== [FAILED] Took 1.11 seconds ==================================================================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  FAILED    00:00:01.110
============================================================================================= 1 failed, 0 succeeded in 00:00:01.110 =============================================================================================El proceso del terminal "C:\Users\egma2\.platformio\penv\Scripts\platformio.exe 'run'" finalizó con el código de salida 1.

Las tareas reutilizarán el terminal, presione cualquier tecla para cerrarlo.

Nbrooks639, 2022-06-09:
i also cannot get these to work, i have the exact same set up but keep getting the same build error

tititopher68-dev, 2022-06-10:
1/ Open PlatformIO Core CLI
Change directory (cd) to the PlatformIO project where platformio.ini is located.
Copy the following pio pkg install command and paste into the CLI shell, press Enter:
pio pkg install --library "adafruit/Adafruit NeoPixel@^1.10.5"

2/ Open platformio.ini.
Add the following line to the lib_deps (after the '=') option of [common:] section:
adafruit/Adafruit NeoPixel@^1.10.5
Build a project, PlatformIO will automatically install dependencies.

Nbrooks639, 2022-06-11:
thank you buddy. 
let me see

Nbrooks639, 2022-06-11:
nope not working

Nbrooks639, 2022-06-11:
[image]

tititopher68-dev, 2022-06-11:
@Nbrooks639
It's a problem of configuration of Platformio and library which does not install....but no firmware since the compilation does not even start...but like that, without having the PC under my eyes, I do not  don't know...

tititopher68-dev, 2022-06-11:
> well no, it doesn't want to load the library :( and it's where it needs to be and everything.
> 
> [image] [image]

Try to rename the folder 'Adafruit NeoPixel' in 'Adafruit_NeoPixel'

Nbrooks639, 2022-06-11:
I use them on all three of my ender 3 s1s just not with this firmware. I like the firmware I just can't use my pixels with it

Nbrooks639, 2022-06-16:
same problem

Nbrooks639, 2022-06-16:
terminal says that the latest version of pixels are installed but it will not put it into the firmware. i even changed the neopixel led menu is the ui dwin.cpp to see if that helped.
crazy i can compile another firmware with them all day long

tititopher68-dev, 2022-06-16:
@Nbrooks639 @ANTABY 
Now it's fixed in [new_feature](https://github.com/tititopher68-dev2/Marlin/tree/new_feature) ! Don't forget to define a Neopixel pin....
It's was me ! Sorry ! I had only removed the redundant call to a leds.h definition file....It was as big as the nose in the middle of the face

Nbrooks639, 2022-06-16:
Now I have another request. Are you able to enable a touch screen via dgus/dwin set?
Doesn't have to run jyers ui just normal functions. And it would not be the s1 pro screen. Ender 3 series touch, cr6 series touch. Theses are both dwin style touch screens

Nbrooks639, 2022-06-16:
now there is problems with the led menu
[image]

Nbrooks639, 2022-06-17:
@ANTABY have you tried since he fixed this? i am able to have the pixels on but the menu option will not work

tititopher68-dev, 2022-06-17:
@ANTABY
What do you mean about "the menu option will not work". 
@Nbrooks639 
There are compilation errors (Typo issues) some of which only depend on what the user defines in the configuration files. I based my compilation tests on configurations for BTT SKR and now all the compilations go to the end. (commit pushed).

Nbrooks639, 2022-06-17:
The errors are in the ui menu nothing I would change. I trying to get the color menu on the printer screen so u can change the colors.  I wouldn't be in the jyers ui changing commas, and semi colons and what not. 
Stm32f103re or stm32f103rc are the boards I use

Nbrooks639, 2022-06-17:
@tititopher68-dev2  I see you fixed. Thank you..

tititopher68-dev, 2022-06-17:
@Nbrooks639 
New commits just now pushed

Nbrooks639, 2022-06-17:
I seen. 
Whoop whoooooop

Nbrooks639, 2022-06-18:
nope something still is not right  it will not build

[image]

Nbrooks639, 2022-06-18:
[image]

tititopher68-dev, 2022-06-18:
@Nbrooks639 
New commit pushed -> Refactor LEDS/NEOPIXEL menus

Nbrooks639, 2022-06-18:
ok im trying it now

Nbrooks639, 2022-06-18:
[image]

Nbrooks639, 2022-06-18:
@tititopher68-dev2 
i dont get what im doing wrong? i am not using a dacai lcd i use the old dwin set lcds idk if that helps
i enable neopixels with pin set they work just fine, just the led menu option is config.h adv is what sets off the errors

Nbrooks639, 2022-06-18:
Wow your right I did the last time but this time I totally forgot

Nbrooks639, 2022-06-18:
alright i finally got it, thank you and sorry for all the trouble @tititopher68-dev2

tititopher68-dev, 2022-06-19:
@Nbrooks639 
No problems ! This makes it possible to move things forward and flush out the issues.  So it's all positive.

Nbrooks639, 2022-06-19:
Yep all good. Working perfect get it gone

ANTABY, 2022-06-24:
> @Nbrooks639 @ANTABY ¡ Ahora está arreglado en [new_feature](https://github.com/tititopher68-dev2/Marlin/tree/new_feature) ! No te olvides de definir un pin de Neopixel... ¡Soy yo! Lo siento ! Solo había eliminado la llamada redundante a un archivo de definición leds.h... Era tan grande como la nariz en el medio de la cara

Thank you very much, now I will try to see if it works. I tried by my means to find the problem for several days, until I gave up. I'm already downloading the latest version and let's try. Thanks for supporting the firmware.