Jyers/Marlin #1951: Question: Inverting Filament Runout Sensor

Kirbo, 2022-05-04:
This is merely a question/request for help:

I'm using this firmware https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin
with our Ender3 v2 + CR Touch.

Now that I tried adding a DIY Filament Runout Sensor, based on the old Z-Stopper switch (like in this article, I also used the same STL's and stuff https://notenoughtech.com/3dprint/filament-runout-sensor-for-ender3-v2/), the sensor works the other way around:
when the filament is loaded, it detects "filament runout" and vice versa.

Is there a way I could invert this from the settings? Or how could I compile the exact same firmware with the `FIL_RUNOUT_INVERTING` inverted or what ever needs to be changed before compiling it again..?

Thanks in advance and keep up the awesome work!

brunohenriquy, 2022-05-04:
In Configuration.h you need to change the line

```
#define FIL_RUNOUT_STATE     LOW
```

LOW to HIGH or vice versa

Kirbo, 2022-05-04:
> In Configuration.h you need to change the line
> 
> ```
> #define FIL_RUNOUT_STATE     LOW
> ```
> 
> LOW to HIGH or vice versa [image]

Thanks for reply :)
That much I understood myself as well, but what other stuff should I define/do to get otherwise identical firmware to https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin ?

brunohenriquy, 2022-05-04:
> > In Configuration.h you need to change the line
> > ```
> > #define FIL_RUNOUT_STATE     LOW
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > LOW to HIGH or vice versa [image]
> 
> Thanks for reply :) That much I understood myself as well, but what other stuff should I define/do to get otherwise identical firmware to https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin ?
> 
> Disclaimer: I have never successfully compiled the firmware. I tried at one point compiling my own, with one setting changed (can't recall which one was it) with no luck. Then I just decided to switch from 3x3 to 15x15 firmware and was happy with it. Therefore I have no idea how to compile in the first place, as it never worked for me.

The equivalent config files for this binary you are using -> https://github.com/tititopher68-dev2/Marlin/tree/new_feature/Configuration%20Files/Ender3V2/UBL-BLTouch-15x15/V422

Kirbo, 2022-05-04:
> > > In Configuration.h you need to change the line
> > > ```
> > > #define FIL_RUNOUT_STATE     LOW
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > LOW to HIGH or vice versa [image]
> > 
> > 
> > Thanks for reply :) That much I understood myself as well, but what other stuff should I define/do to get otherwise identical firmware to https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin ?
> > Disclaimer: I have never successfully compiled the firmware. I tried at one point compiling my own, with one setting changed (can't recall which one was it) with no luck. Then I just decided to switch from 3x3 to 15x15 firmware and was happy with it. Therefore I have no idea how to compile in the first place, as it never worked for me.
> 
> The equivalent config files for this binary you are using -> https://github.com/tititopher68-dev2/Marlin/tree/new_feature/Configuration%20Files/Ender3V2/UBL-BLTouch-15x15/V422
> 
> Take these files and move them to the Marlin folder, replacing the ones there.
> 
> To compile, check any youtube tutorial, since you have these correct files, compiling it will be easy!

Okay, will try that out. Thanks! ☺️

tititopher68-dev, 2022-05-04:
@Kirbo @brunohenriquy 
If you want use my configuration files, you MUST use only my sources (https://github.com/tititopher68-dev2/Marlin/tree/new_feature), not others sources ! You have to compile your own firmware with these sources.

You cannot mixing my configuration files with others sources like Jyers v2.0.1 ! My configuration files have a lot of modification for the runout sensor. With my Sources, you can disable, change the type of runout sensor (Switch mode LOW or HIGH, Motion Sensor like BTT SmartSensor) in the UI without need to recompile !

brunohenriquy, 2022-05-04:
> @Kirbo @brunohenriquy
>
> If you want use my configuration files, you MUST use only my sources (https://github.com/tititopher68-dev2/Marlin/tree/new_feature), not others sources ! You have to compile [...]

I know, the link I provided him is from your fork. 😊

Kirbo, 2022-05-05:
I’m not 100% sure how I feel about others advertising their own fork on someone elses..
The initial feeling is ”that’s kind of fishy” or dodgy or something the least. As if Apple was promoting their stuff on Debian’s repository or similar. That doesn’t feel right.

I like Jyers, and on Jyers I will stay. I’ll findout myself then how to configure and compile it, rather than jump on a ”neverheard brand” instead.

No offense, but I wasn’t asking ”which fork should I use”, I asked ”how to compile THIS fork”.

adam3654, 2022-05-05:
Jyers hasn't been very active as of recent and tititopher68-dev has stepped up to continue the project. He's not trying to steal "customers" or anything. There has been a lot of improvements made in the `new_feature` branch linked above

brunohenriquy, 2022-05-05:
> I’m not 100% sure how I feel about others advertising their own fork on someone elses.. The initial feeling is ”that’s kind of fishy” or dodgy or something the least. As if Apple was promoting their [...]

I am sorry, I should have provided the context why I pointed to @tititopher68-dev's fork.
Anyway, take a look here. -> https://github.com/Jyers/Marlin/tree/v2.0.1/Configuration%20Files/E3V2%20Templates

tititopher68-dev2, 2022-05-05:
> I’m not 100% sure how I feel about others advertising their own fork on someone elses.. The initial feeling is ”that’s kind of fishy” or dodgy or something the least. As if Apple was promoting their [...]

@Kirbo 
When I read this, I really want to tell you to stay on the V2.0.1 but you should not come and complain about the outcome that you will meet afterwards and it is unfortunately very common, if you had been reading the  Different posts in the discussion section, you would have learned that I took up the development of the project following the absence of Jyers, while keeping it informed of what I did on Discord in private message.  I think that without that, the project would be dead today.  Finally Jyers and I, let's collaborate together even if for the moment it is not available.

tititopher68-dev2, 2022-05-05:
@everyone Link to learn how to compile a firmware... in French : https://github.com/fran6p/Documents-LI3D/blob/main/Comment%20installer%20VScode+AutoBuildMarlin+Platformio.md

Kirbo, 2022-05-05:
> > I’m not 100% sure how I feel about others advertising their own fork on someone elses.. The initial feeling is ”that’s kind of fishy” or dodgy or something the least. As if Apple was promoting their stuff on Debian’s repository or similar. That doesn’t feel right.
> > I like Jyers, and on Jyers I will stay. I’ll findout myself then how to configure and compile it, rather than jump on a ”neverheard brand” instead.
> > No offense, but I wasn’t asking ”which fork should I use”, I asked ”how to compile THIS fork”.
> 
> @Kirbo When I read this, I really want to tell you to stay on the V2.0.1 but you should not come and complain about the issues that you will meet afterwards and it is unfortunately very common, if you had been reading the Different posts in the discussion section, you would have learned that I took up the development of the project following the absence of Jyers, while keeping it informed of what I did on Discord in private message. I think that without that, the project would be dead today. Finally Jyers and me, let's collaborate together even if for the moment it is not available. I seek neither glory nor light. I'm keeping the project alive for now and while Jyers is away, for this community and for those who have adopted the Jyers project!

Didn't mean to complain or raise any issues, I was merely asking for help/guidance on "HOW TO".
I wasn't aware either that Jyers is not actively maintaining this repo anymore nor that someone else has "taken it under active maintenance on their own fork" (I've been into 3D printing scene only for 2 months for now and everyone everywere have been praising Jyers, therefore I chose to use it and that's also the reason I reacted the way I did once I noticed that someone else is promoting their fork here).
But the way I was promoted to jump in a train of another fork without explicitly mentioning "you should perhaps use this one instead", rather just giving me a link to a whole another fork (yeah, I know, I should've read it better myself as well).

I will give your fork a try, now that I've had time to read everything on a computer, instead of quickly scrolling through stuff on a phone.

And no, I have not read the discussions either, my bad. 

Sorry if I caused any annoyance or anything, it wasn't intentional.

But for the "I'm keeping this alive while Jyers is away" part, I think it would be best to have that said in the README.md file in this repository, preferably even the first line (or at least at the very top of the file) so that anyone opening this repository would notice that "okay, so this repository isn't under active development anymore and the author of this repository suggests others to use this one instead".
I don't know, just my 2 cents.

Kirbo, 2022-05-05:
> @Kirbo @brunohenriquy If you want use my configuration files, you MUST use only my sources (https://github.com/tititopher68-dev2/Marlin/tree/new_feature), not others sources ! You have to compile [...]

Will definitely give this one a try, especially if I don't even need to compile anything myself ❤️

closed 2022-05-05 by Kirbo