Jyers/Marlin #1956: Been unable to have a succesfull print since installing 2.0.1

larie438, 2022-05-08:
I installed version 2.0.1 on my ender 3v2 with 4.2.2 board and since then I have been unable to have a successful print.
I'm using a CR touch with a HS 10x10 UBL mesh and no matter what I do prints either warp tremendously or they don't stick at all.

I have adjusted z offset, yx offset, and leveled the bed, adjusted the e steps. nothing works

this is my starting G code

G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 L0 ; Load UBL Mesh at Slot 1
G29 A ; Activate UBL
G29 J2 ; Tilt the Mesh

tititopher68-dev2, 2022-05-08:
@larie438 
Do a PID auto-tune. What kind of filament are you printing?  Warping and adhesion depend on multiple factors.  The Zoffset but not only that.  But also lots of parameters in the slicer.  Such as initial layer height, bed temperature for first layers, cooling, first layer crush rate, first layer print speed, bed cleanliness.  Warping is generally due to too rapid cooling of the first printed layers below the crystallization point, creating internal tensions.  The larger and more angled the part to be printed, the more warping it will be.
 Regarding your calibrations, bed leveling, Zoffset, build mesh, etc... always do them at printing temperature (and also not immediately once the temperatures are reached, wait a bit to standardize the Bed temperature...).  Also do a temperature tower with your filament to really determine the right printing temperature. Placing a printer in an area where it can get light drafts can be fatal to print quality.

larie438, 2022-05-08:
@tititopher68-dev2 I have done the PID auto-tune but did not help. before installing 2.0.1 I was having success with the stock firmware, I'm sure the print quality could have been improved but I was able to print fairly nice parts, now I can't even print a simple benchmark.
for all my print I'm using PolyTerra PLA +

deathlessdd, 2022-05-09:
> @tititopher68-dev2 I have done the PID auto-tune but did not help. before installing 2.0.1 I was having success with the stock firmware, I'm sure the print quality could have been improved but I was [...]

I had the same kinda issues as you once I upgraded firmware I even got dual Z axis upgrade I upgraded the bed twice. I went to PEI bed and i’ll never go back to glass ever again It’s almost a perfect print every single time.

deathlessdd, 2022-05-09:
I had the same kinda issues as you once I upgraded firmware I even got dual Z axis upgrade I upgraded the bed twice. I went to PEI bed and i’ll never go back to glass ever again It’s almost a perfect print every single time.