Jyers/Marlin #1959: Buzzer vs speaker for Ender

andycarver101, 2022-05-11:
Do you have plans to enable the speaker and replace the buzzer in any of your releases?

my current install only has one tone 

super excited for the m300 command..

tititopher68-dev, 2022-05-11:
@andycarver101 
Sorry but not possible without hardware modification on the motherboard.

andycarver101, 2022-05-12:
> @andycarver101
> I just saw you go to the "Magician" who recovers the Jyers code for the UBL and who hides part of his code falling under the GPL and in this does not respect the GPL! So I hope for you that the "Magician" can via his firmware solder you on the 2-pin motherboard and a remote speaker...
>
> Rares sont les cartes mères Creality V4.2.x veant avec au moins les pins ici déjà soudés:

Hi,

I’m assuming you mean,
https://github.com/mriscoc

Thank you for the detailed answer. no one has been able to provide with this much clarity. 

I’m sorry people are abusing your code. 

I have not yet committed to this firmware as it has not been tested by me. 

I moved as it seemed more up to date. 

I hope you two can resolve your situation. 

Andy Carver
***@***.***

tititopher68-dev2, 2022-05-12:
@andycarver101 
Just for information, following the (too long...) absence of Jyers and while waiting for his return, I resumed the development of this project here: https://github.com/tititopher68-dev2/Marlin/tree  /new_feature (you will have to compile your own firmware, in the Discussion area, I posted a link to an image tutorial).  Jyers is of course informed of all progress.
 There is a Discord whose link is in #1694.
 There is also another Discord from a Youtuber friend (but in French).
 The version of the link is v2.1.0a....But my final version v3.0.0a will be released soon, which will have more possibility than that of the other Dev with a 100% open code and without a line of code  hidden.  The version is ready, I am still preparing all the documentation.  Only a very small number of people had access to v3.0.0a to look for issues.  I had to do this because this dev, under other nicknames, comes to do his shopping.