Jyers/Marlin #1962: Heating up issue

Heatley2001, 2022-05-15:
I've downloaded and installed the newer version of the default 4.2.2 firmware for me V2.

The issues im running into is when I try and heat my nozzle up to 220 like I was doing with the original ender firmware.  With the new firmware it will reach up to 205 then give me a heating failed error please restart.  I restarted and the same happened again.

I did the PID auto tune and it will heat up to 220 with no problems. It only gives me an error when I go back to the main menu and preheat like normal.

Im also running into an issue when I auto home it. After it goes back to the home spot and I preheat it it will once again give me a heating error.

The only time it won't give me a heating error is when I disable steppers and raise the nozzle up a small bit from the bed. Then it will heat all the way up with no error. As long as I don't choose auto home I can choose what print file I want and it will heat up and start printing like normal. Just as long as I don't have it auto homed.

I'm really scratching my head on what to do next. Any help would be great.

Thank you.

tititopher68-dev2, 2022-05-15:
@Heatley2001 
Modifié in Configuration_adv.h #define THERMAL_PROTECTION_VARIANCE_MONITOR ->  //#define THERMAL_PROTECTION_VARIANCE_MONITOR and compile your onw firmware.
This experimental functionality introduced by Marlin give a lot of issue.

Heatley2001, 2022-05-15:
@tititopher68-dev2  I'm not sure how to compile my own firmware. Is there a way to edit the existing firmware for that option?

tititopher68-dev, 2022-05-16:
@Heatley2001 
Sorry but not option to edit a compiled firmware. But you have a Tuto here : #1953