Jyers/Marlin #1965: Cannot compile any firmware

stevemcc1980, 2022-05-29:
### Description

Been trying to increase my maxtemp to 285 using bltouch 3x3 ender 3v2 configuratio files but i cannot get it to build

### Steps to Reproduce

**Expected behavior:** Firmware Compiles

**Actual behavior:** Error - Processing STM32F103RE_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------Error: Could not find one of 'package.json' manifest files in the package

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Lothaedan1, 2022-06-02:
Hi there!

I may be no expert, but it sounds like the build enviroment is incorrect.

go to the .ini file and maybe try switching the "default_envs" to something else. Maybe mega2560?

stevemcc1980, 2022-06-02:
Thank you for your reply!
I’ll maybe try that. I’m not sure why the crealityv4 is causing issues. Like you said may be something missing

stevemcc1980, 2022-06-14:
Hey i still cannot get ANYTHING to compile. No matter the default_envs i get the package.json error. 

Any help would be greatly appreciated

dmerriman11, 2022-07-01:
Change default_env to STM32F103RET6_creality, that should do the trick.