Jyers/Marlin #1967: Homing with Bltouch and Z end switch as a backup

zaggash, 2022-05-31:
### Description

I'm using the latest [E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin](https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin)

During the Homing, the Z-axis is stopped with the help of the BLTouch which is good and expected.
However, I tried to manually press the Z End Switch manually during Homing but it is ignored.

What if the BLTouch fails ? or doesn't work correctly?
The Nozzle would hit the bed.

Is there a way to prevent that and make Bltouch and Z end switch as a backup?

Feels like a bug to me since we have a dedicated port on the motherboard for the BLTouch

### Steps to Reproduce

1. Flash with [E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin](https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin)
2. Home the Printer
3. Wait for the Z Homing.
4. Force the Z stop with the Z End Switch

**Expected behavior:**

The Printer should stop the Homing session and raise the Z 

**Actual behavior:**

It does nothing.

zaggash, 2022-06-01:
The BLTouch replaces the Z-Endstop if you wire it into the Z endswitch on the motherboard.
Why is there a way to wire both BLTouch and Z endswitch on the motherboard separatly then?

I stumbled upon this convo and it gets closed because of inactivity.
https://github.com/MarlinFirmware/Marlin/issues/15731
I wonder if it would be a nice addition to the community firmware.

I know it is on the configuration.h but it doesn't mean it is not possible to make it work with a bit of configuration knowledge on Marlin.

amandamana, 2022-06-02:
There are some good arguments in that linked convo, and another convo linked within it. I'd love to set my Z-Endstop 1.5 mm too low and use it to protect my bed in case of a probe failure.