Jyers/Marlin #1968: In the UI, enabling the "Filament Sensor" option does nothing to actually detecting and using the filament sensor attached.

Lothaedan1, 2022-06-02:
### Description

Title says it all. My filament sensor turns the indicator light on and off accordingly, but the firmware does not seem to really care.

### Steps to Reproduce

1. Turn on Filament Sensor option
2. Install Filament Sensor
3. Be amazed at how nothing happens

**Expected behavior:** The Filament sensor should work

**Actual behavior:** It doesn't

#### Additional Information

This was using the official release bin file you put out. I tried creating my own to solve the problem but ran into an even bigger problem. Something along the lines of the build environment being incorrect, but if I change it to mega2560 like it asks it tells me that my "'ENCODER_PHASE_0' was not declared in this scope". Weird. This is of course using the unmodified base file.

tititopher68-dev, 2022-06-02:
@Lothaedan1 
Hardware issue, check your wires, please !

ruettenc, 2023-02-11:
i have a similar issue with the sensor in that it initially works but every time i try to print, it says "filament runout" then purges filament, then gets stuck on "filament runout" again and i have to reboot to clear it.