Jyers/Marlin #2011: Abandoned?

perelin, 2022-08-19:
Is this repository abandoned? Does development continue elsewhere?

bellertm1, 2022-08-20:
**Bump   Same question.

Ergonomicmike, 2022-08-20:
So, a Sanity Test: What does the jyers version bring to Marlin, now that Marlin is at 2.1?

I'm a totally newb, but I saw some code from jyers in the Marlin code.

And I just built Marlin 2.1.x for my Ender and saw stuff about Manual Mesh leveling in the configuration.h file. 

I had flashed my display for jyers and can't see much on it now with Marlin 2.1.x running (I will have to 'unflash' it). But are there still things in jyers that one cannot do in Marlin 2.1?

Ergonomicmike, 2022-08-21:
How will this code differ from what appears to be a similar project at
https://github.com/LCh-77/Marlin/tree/bugfix-JyersUI ?