Jyers/Marlin #2012: I was able to build it for Ender 3 V2 4.2.2.

Ergonomicmike, 2022-08-22:
I know that this isn't the correct fork in which to make a comment about a similar fork. But since there's no way to comment in that other fork, I want to give some encouragement to @LCh-77. After four days of hacking (and I mean 'hacking' - numerous WinMerge comparisons with old Jyers, almost breaking a belt because I didn't have a stepper motor direction reversed in the config.h, etc.) I was finally tripped across this particular version of Jyers/Marlin and was able to build an updated Jyer's Marlin v2.1.x using the code from https://github.com/LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Update.

It looks like the 3x3 Mesh version that I had before from the old Jyer's bin, except that the icons aren't painting. (If I had to guess, as someone who doesn't know code, I am guessing that the icons aren't enumerated as they are in the old jyer's dwin.h.)
[image]

LCh-77, 2022-08-22:
Ah that is because you must to write #define USE_STOCK_DWIN_SET, if you have the stock icon set,

Ergonomicmike, 2022-08-22:
Thanks for the quick reply.

I am using the Jyer's Gotcha icons. They worked fine with his recompiled Manual 3x3 bin. And I saw them appear briefly when I flashed your code onto my motherboard while still having TH3D's firmware on my display. It was only after I flashed my Display back to Creality (using the Jyers DWIN) that the icons disappeared. (There's also a strange problem scrolling thru the "pages." Better shown in a video.)

I think I tried the "Enhanced" branch. I couldn't get it to build. (Errors about "Row1" not being declared. Stuff way over my head.)

Ergonomicmike, 2022-08-22:
P.S. I'm willing to try the "Enhanced branch." But with so many branches here (I only found your -Update branch by mistake), can you please give me the link?

Else, I can wait for your big Pull Request to comit on the Jeyers branch.

Ergonomicmike, 2022-08-22:
I just downloaded the latest Enhanced version, which was updated a few hours ago (Thanks.) 
https://github.com/LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced

I builds to completion when I select the DWIN_CREALITY_LCD in configuration.h

I flashed my Ender 3 v2 v4.2.2 board and the display icons now show, albeit I don't have the Mesh stuff in the 4th placeholder when "Info" normally shows.

But when I select DWIN_CREALITY_LCD_JYERSUI I receive the following error nearing the end of the build. (It is the ONLY change to the .h file.)

```
Marlin\src\lcd\e3v2\jyersui\dwin.cpp: In static member function 'static void CrealityDWINClass::Menu_Item_Handler(uint8_t, uint8_t, bool)':
Marlin\src\lcd\e3v2\jyersui\dwin.cpp:3933:25: error: 'row1' was not declared in this scope; did you mean 'row'?
 3933 |               sprintf_P(row1, PSTR("%ix%ix%i"), eeprom_settings.x_bed_size, eeprom_settings.y_bed_size, eeprom_settings.z_max_pos);
      |                         ^~~~
C:\Users\20H2\.platformio\packages\framework-arduinoststm32\cores\arduino/avr/pgmspace.h:71:36: note: in definition of macro 'sprintf_P'
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
      |                                    ^
*** [.pio\build\STM32F103RC_creality\src\src\lcd\e3v2\jyersui\dwin.cpp.o] Error 1
```

LCh-77, 2022-08-23:
I  posted a release: https://github.com/LCh-77/Marlin/releases/tag/BLTUBL-LA
Also now the Enhanced branch comes with working configuration files.

Ergonomicmike, 2022-08-23:
Yep, I saw the new release and updated configuration files. The build errors did not pop up this time - and this was still with the JyersUI selected.

Many thanks for your hard work. (I've been at this for 3 or 4 days, hacking. I can only imagine what it must be like for you all.)

closed 2022-08-24 by Ergonomicmike