Jyers/Marlin #2015: bugfix-2.1.x_JYERSEI-Update on Ender 3 v2 4.2.2 success! (Almost)

Ergonomicmike, 2022-08-23:
This morning I tripped across branch bugfix-2.1.x_JYERSUI-Update (https://github.com/LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Update), 
which appears to be the long awaited comit'd version of  https://github.com/Jyers/Marlin of this branch.  (I don't understand how github branches work and why the https://github.com/Jyers/Marlin/ here isn't updated with the fixes that are in -Update. Once things settle down with this Jyers update, it would be nice if someone went through and deleted the offshoot branches.) 

Using the new code in -Update, I built firmware for my Creality Ender 3 v2, selecting the JyersUI in the configuration.h.

(Of course I had to carefully go through the OEM Configuration.h and Configuration_adv.h, comparing them to the latest example configurations for the Ender 3 v2 (at https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x) and change as needed. (If you don't, your stepper motors will run backward.) And even after you change the OEM .h's, you will see a few errors during the build which fortunately the Build tool suggests corrections. (E.g., enable SOFT-PWM))

Success! Whereas yesterday, the JyersUI did not display correctly.

I had to play some tricks to get the display to work. (As described below). But it looks perfect so far. Almost.

(Actually, I'm so excited that I haven't run my printer through its paces yet. I've only played with the Display. But it homes okay. So at least I have the true/false settings for the steppers set correct now.)

As you can see in the image, attached, there appears to be a z-order problem with the background bitmap. You can see the Creality home page image poking through in four places.
[image] Not a show stopper.

Now, some tricks that I had to do so that the Display would work.

First, you HAVE to use the latest display files from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x

In the case of the Ender 3 v2, config > examples > Creality > Ender-3 V2 > LCD files > private.

It's the 'private' that you want, because this new 'private' contains the jpg's for the Jyers' UI. (Like the multi-colored mesh icon.)

(Interestingly, there are two of each icon. I don't know why.)

So I'm guessing that if you had Jyers working before with the Nov 2021 version, as I did, that you will have to update your Display with "private' for this new build to display icons correctly. 

Simply copy 'private' to your SD card, put the SD card in the Display, power up and watch it load the icons into memory.

EXCEPT - if you have Display board version 1.2, as I have (and perhaps higher), it appears that Creality cheaped out on memory. Because I could not load all the icons. (I received an error about 7/8 of the way through.) I recall seeing something about 2MB of memory when the update process first started. Which should be enough for all the icons. But what do I know?

So I had to delete the 3.zico files from private in order for the update process to complete. (I also changed the Marlin logo jpg to one background color to shrink the size of that jpg.) After I made these changes, the files in 'private' loaded into the Display.

These 3.izco icons appear to be used for a non-Jyers' (that is, a vanilla Creality) UI. And so the 3.zico files would appear in the fourth placeholder, titled "Info" in the Creality UI. But of course, this placeholder is used for Mesh in the Jyers' UI and not Info.

If the duplicating of every icon in 'private' was unintentional (i.e., the dupe icons are not needed), then the 'private' folder would work if the dupes were not there, since the memory required for all the icons would shrink by almost half.

LCh-77, 2022-08-23:
I  posted a release: https://github.com/LCh-77/Marlin/releases/tag/BLTUBL-LA
Also now the Enhanced branch comes with working configuration files.

Ergonomicmike, 2022-08-23:
Yes I saw the Release. And I saw the updated configuration files. (Although I still think that the motherboard should be defined as BOARD_CREALITY_V422. (But I also saw your warning during the build about defining the correct Stepper Drivers.))

Very thorough documentation - I like the settings that you specified as Ender specific. (I had figured out most - but not all - of them from hacking Jyers' earlier configurations.)

I still had to build my own firmware because I use the 3x3 Manual Mesh (which works quite well, and I don't expect that I'll ever have to buy a BL Touch.) But it built fine this time with the JyersUI and I didn't select USE_STOCK_DWIN_SET. (I'm using the previous Jyers' icon set. (After @ellensp had re-compressed the icons for me.))

Thanks! It's nice to have an updated Jyers.

ellensp, 2022-08-23:
repacked jpegs from marlin config/examples/Creality/Ender-3 V2/LCD Files/private/  can be found here https://github.com/MarlinFirmware/Configurations/pull/807/files  until they get merged.

LCh-77, 2022-08-24:
> Thanks! It's nice to have an updated Jyers.
You are welcome, that is in fact an updated Jacobs firmware (not my) and it will be always Jacob's baby.

closed 2022-08-24 by Ergonomicmike

Popeye8564, 2023-01-03:
I just downloaded marlin for screen upgrade the screen turned red or orange right away. screen didnt  change