Jyers/Marlin #2016: What is the different between bugfix-2.1.x_JYERSUI-Update and -Enhanced?

Ergonomicmike, 2022-08-23:
Not an issue, but a question. The ReadMe's don't make it clear, except to say that the -Enhanced is based on original Jyers? Yet the -Update has the name "Jyers" in it too.

The files look mostly the same from a time stamp/file size perspective. (Except that Enhanced has a few more commits to it.)

I didn't see anything different in their Configuration files.

While I was able to build the -Updated code without error, I was not able to build the -Enhanced. 

After getting a warning about a problem with dwin.cpp in -Enhanced, I hacked by copying the 10 minute newer dwin.cpp from -Update to -Enhanced. (Developer rolls eyes.)

And after that, an error in gcode.h. So I copied the file from -Update to -Enhanced, and it completed.

tititopher68, 2022-08-23:
I invite you to join our Discord Server (https://discord.gg/vDJQQuKM).  You will find there the real source code of the firmware (much more up-to-date by the way) of which LCh-77 has shamefully and dishonestly claimed paternity and ownership.  He didn't code anything, just copied my code (which contains the identification markers!).  He is therefore unable to provide real support and fix bugs... he only copied himself.  He will certainly defend himself by pretending and believing his own lies.  Feel free to join us as almost the entire community here did.  I am in contact with Jyers.  He encourages me to continue my developments and assured me that he will obviously not merge the PR of LCh-77.

HAPSagan, 2022-08-23:
Hello
The problem for us who don’t speak French is that we don’t undestand where to find the infomation and can’t read the instructions. 😢
Best regards, Peter

LCh-77, 2022-08-24:
> I invite you to join our Discord Server (https://discord.gg/vDJQQuKM). You will find there the real source code of the firmware (much more up-to-date by the way) of which LCh-77 has shamefully and [...]

You start over, who copies code from everywhere is you! That's why your account was closed and now you can't publish your code in Github to avoid to be compared. I saw that you also copied all the improvements I made to JYers code and distributed the firmware as your own. But, your copy is so bad than you are mixing the same functions from different sources duplicating code. Your are not only a plagiarist but it seems like a bad programmer too.

tititopher68, 2022-08-24:
@LCh-77 ahahahahahah ! A fine example of your mythomaniac narcissistic pervert physicality.  Come on, I challenge you: in my code, I put recognition markers.  Among these markers, there is my daughter's first name.  How is it that we find this name in the same place in your code?  Because you copied my code!  I'm not going to answer you anymore because it feeds your perversion.  Everyone knows and knows your misdeeds.  You can continue your delusions as much as you want if that relieves you.

LCh-77, 2022-08-24:
**Ttph68** Any real programmer can inspect your code and will be horrified of how obvious it is that you copy code from others, multiple code styles, duplicated functions, unoptimized code, etc, etc. All my replies will be technical, personal attacks are reserved for people without arguments.

closed 2022-08-24 by Ergonomicmike