Jyers/Marlin #2017: Small Display nit with fonts (using JyersUI)

Ergonomicmike, 2022-08-24:
So first, I am running a build based on the latest version of bugfix-2.1.x_JYERSUI-Enhanced but with the JyersUI selected in configuration.h. But the instructions for using this code base is to use the DWIN_CREALITY_LCD. So I realize that I am running an "unsupported" version of the code at this time.

But I had the same problem when I ran the more official Ender3V2_422BLT_UBL_LA.bin too. And since the font used in DWIN_SET is the same as what I'm using now, I expect that this is a small screen refresh problem in a Display 'subroutine' (? FORTRAN days) somewhere.

### Description

There are two problems, perhaps related:

1) The font for Z-Offset is not being cleared/refreshed on rewrite. And so in this photo, you are seeing both an initial zero and a five overlaying each other. It might be that the font for the Fan will have the same problem, as I can see some breaks in the zero.

2) Similar issue with the smaller fonts at the bottom. They appear to be an overwritten mess. During printing, they turn into a big white blog.

### Steps to Reproduce

In Configuration.h, set DWIN_CREALITY_LCD_JYERSUI (and comment out the other choices) and use the icon pack from here:
https://github.com/MarlinFirmware/Configurations/pull/807 for the Creality LCD Display.

**Expected behavior:**

The fonts displayed correctly in the November 2021 release of Jyers-bugfix, 

[image]

tititopher68, 2022-08-24:
@Ergonomicmike 
This is what happens when LCh-77 copies code without mastering it...

LCh-77, 2022-08-24:
For use JYersUI you must use DWIN_CREALITY_LCD_JYERSUI, I don't have that flaw in the text background, now it is transparent to allows us to use different background colors. Be sure to use Stock, the `DWIN_SET` or `privated` display firmware published in my repository: https://github.com/LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced/display

Ergonomicmike, 2022-08-24:
You are correct. It worked. (Although, as you know, the cute Jyers' mulitcolored math mesh icon for leveling isn't there anymore.)

I apologize for not first doing what you said to do in your Instructions, before creating this (non)-Issue. I didn't think that your 'private' set could work with the JyersUI without having those jpgs that were in earlier versions. But I learned last night, after trying the Frenchman's 'private,' that icons can be stored in a .zico file.

So everything is fine now.

Now I guess I will have to learn all about Linear Acceleration (if it works with my A4988 drivers).

closed 2022-08-24 by Ergonomicmike