Jyers/Marlin #2018: Print Confirmation dialog appears even though commented out

Ergonomicmike, 2022-08-24:
### Description

IIRC, I never had to confirm the start of a print. But I do now, despite 
` // #define SD_MENU_CONFIRM_START             // Confirm the selected SD file before printing`
in configuration_adv.h.

Could it be another dialog being triggered because I don't have a thumbnail for this print?

### Steps to Reproduce

Print a file from the SD card that does not have a .jpg thumbnail associated with it.

**Expected behavior:**

Since I have commented out "Confirm the selected SD file before printing," the print should start without asking for confirmation.

**Actual behavior:**

But it doesn't. See photo.

#### Additional Information

I'm guessing that it has something to do with the lack of a thumbnail now, a new feature in this build? 

I did a quick search of the github and I did find a file that mentions 'thumbs." But that snippet of code didn't appear to be for the Ender LCD.

I suppose I should figure out how to add a thumbnail to the gcode in Cura and test my hypothesis.
[image]

Ergonomicmike, 2022-08-24:
Just now I saw this update to something called a mriscoc codebase.
https://github.com/mriscoc/Ender3V2S1

`- Custom G-code C10 to mark a G-code file as a configuration file to avoid confirm to print and print done dialogs.`

tititopher68, 2022-08-24:
@Ergonomicmike 
I have the answer because I am the real custodian of this code but I await with delight that the urupator manifests itself to give you a real answer that holds water....As I already told you, join us on  the Discord and you will have all the help and answers possible (even if it is in French, with a translator no problem. You can ask your questions in English too if you want).

LCh-77, 2022-08-24:
SD_MENU_CONFIRM_START  is only taken in to account in MarlinUI, I don't implement that yet in JYERSUI.

Ergonomicmike, 2022-08-24:
"SD_MENU_CONFIRM_START is only taken in to account in MarlinUI, I don't implement that yet in JYERSUI."

Okay, thanks. A comment in the .h that it only affects MarlinUI would be helpful.

LCh-77, 2022-08-24:
In VS code, you can use the search function to review where a value is used.

Ergonomicmike, 2022-08-24:
Okay. I just learned about VS Code a few days ago. Still overwhelmed and barely able to build. But if there's a next time, I'll try to sniff around to see what affects what. 

I can't remember if I always had to press "Confirm" with the November Jyers release. I suppose I'll have to flash back to that for a moment to see.

closed 2022-08-24 by Ergonomicmike

mriscoc, 2022-08-24:
> I can't remember if I always had to press "Confirm" with the November Jyers release. I suppose I'll have to flash back to that for a moment to see.

Confirm comes with the new feature to show g-code previews.

I use the code from Marlin to show the previews, looks like it follow this convention: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview

mriscoc, 2022-08-24:
> I used some of @mrisco code (With the proper autorship citation) for some of the custom gcodes in JYERSUI, C10 is a recent addition, I don't know if I will to use it.

You can use freely my code always that respect licences and authorship.

mriscoc, 2022-08-24:
> I used some of @mriscoc code (With the proper autorship citation) for some of the custom gcodes in JYERSUI, C10 is a recent addition, I don't know if I will to use it.

You can use my code freely as long as you respect the licenses and authorship.