Jyers/Marlin #2019: What does "NaN" mean on the Mesh page?

Ergonomicmike, 2022-08-24:
Running a build of the latest code base of bugfix-2.1.x_JYERSUI-Enhanced. I set the build to use the JYERSUI, and I have the JyersUI icon pack installed in my Display.

### Description

When I am performing Mesh Leveling for the first time (Manual leveling, 3x3 Mesh. no probe), the Z Value at the bottom of the page shows "Nan." (See photo.) 

Not a big deal - it changes to a number after the Mesh has been saved. 

### Steps to Reproduce

Compile bugfix-2.1.x_JYERSUI-Enhanced with Configuration.h set for Bilateral leveling, Manual Probe, JYERSUI and install the Jyers' icon pack in your display.

**Expected behavior:** 

Z Value should show zero initially.

**Actual behavior:** 

It's shows "Nan."

Should it be dashes? Or zero?

#### Additional Information

It's really a small "nit." (Pun intended: "Nit" - "Display" Get it?)

[image]

LCh-77, 2022-08-24:
NaN means that point was undefined, due to invalid mesh, probe failure, unreacheable point, etc.

LCh-77, 2022-08-24:
NaN = Not a number

Ergonomicmike, 2022-08-24:
Interesting. I didn't know that that was a "thing." But then, I'm not a coder.

I still think that dashes would be more ergonomic, and more consistent with the dashes at the bottom of the Home Page when there are no numbers.

closed 2022-08-24 by Ergonomicmike