Jyers/Marlin #2022: 5x5 BL Touch mesh doesn't save after doing a manual tuning?

TrevKo5000, 2022-08-24:
After creating a 5x5 mesh by probing with the BL Touch, the mesh will save. After doing a manual tune to the current 5x5 mesh, going point by point and then pushing "Back", the manually tuned mesh will then be shown when I select "view mesh". However, after turning the 3D printer off and then on again, the manually tuned mesh will then revert back to the original probed 5x5 mesh. 
Is there a way to save that manually tuned mesh?

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

LCh-77, 2022-08-24:
The printer uses the mesh in RAM, EEPROM is only for storage and retrieval, after editing your mesh you have to save it for future use.

Ergonomicmike, 2022-08-24:
To be a little more complete in the answer, after tuning your mesh, you need to scroll over to the Control icon on the "home" page, drill down to "Store Settings" and press the knob. (Two beeps.)

One would think that the new mesh would automatically store after tuning the mesh. Or that the option to save would be on the same page as "Mesh Edit," as it is for "Create Mesh." Not as ergonomic as I would like.