Jyers/Marlin #2023: Stuck in a Loop ater cancelling a print

Ergonomicmike, 2022-08-25:
### Description

LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced

BLTUBL-LA/Ender3V2_422BLT_UBL_LA.bin

When you Stop a print, you cannot return to the Print Menu. (In fact, IIRC, even when the print is finished, I am stuck in a loop and have to power down the printer.)

### Steps to Reproduce

1. Start a print. (Hit Confirm.)
2. Before the print is finished, Stop the print. (Hit Confirm.)

**Expected behavior:**

Expect to return to the Print Menu.

**Actual behavior:**

Stuck in a loop, per the video below.

#### Additional Information

I made a 2 minute video to show the issue.

I first start out with my own build, where, per the OEM Configuration.h file, both JYERSUI and USE_STOCK_DWIN_SET are defined.

You can see the loop.

Then, as an experiment, I compiled another build where I uncommented DWIN_CREALITY_LCD_JYERSUI

In that segment, you can see that this time, a message pops up after you stop a print, telling you to wait while the head parks.

Then you return to the Print menu.

(You can also see that, without the JYERSUI component, there is no "Confirm" to print.)

Last, I flashed BLTUBL-LA/Ender3V2_422BLT_UBL_LA.bin so that everything is vanilla.

The problem is there too.

https://streamable.com/07o067

tititopher68, 2022-08-25:
@Ergonomicmike 
This is another demonstration of what happens when this guy who claims to be a great developer actually shows that he sucks.  He made a dodgy mix of code from Mriscoc and my code.  He says that I suck, which is not at all the opinion of Master Jyers himself (who I repeat wishes to use my code).  At home and for users of my firmware (the real one!), we have none of your issues....Ah that's right, LCh-77 boasts of having super optimized code.  We see the result Ah ah ah... But you have chosen your side so manage on your own now, you see what I mean with your behavior on the Discord...

LCh-77, 2022-08-25:
Thank you for finding the error, which allows us to improve Jacob's code. At first I couldn't confirm it, but later I saw that you're not using preview images. That was the difference, I had a mistake in the PrintDone confirm screen. It is now fixed.

LCh-77, 2022-08-25:
> @Ergonomicmike This is another demonstration of what happens when this 

[image]

It is not your firmware, it is Jacob's firmware, "A thief believes everybody steals"

Ergonomicmike, 2022-08-25:
Confirmed here too. Thank you for fixing this so quickly. I am a beta-tester for an Aviation software company and I have found that making videos to demonstrate a problem helps the developers to see what's wrong.

Now that you found that this issue was related to not using Preview images, I wonder if the same is true with the "Confirm" to print dialogue? I can see how a developer might want users to confirm their choice of file if they can't see their choice.

If I knew how to code, I would investigate this for myself and change it locally.

Also thanks to you all for fixing - in your Release - the long-standing error in Mr. Meyer's Configuration.h that had incorrectly set the bed size for the Ender 3 v2.

closed 2022-08-25 by Ergonomicmike

tititopher68, 2022-08-25:
@LCh-77 
Jacob will never do anything with you. He assured me. On the other hand, he thanks me for all the work done and encourages me to continue my developments regardless of your delusions. Keep feeding your mythomania. Just know that we will demonstrate everything in a YouTube video (in French and in English) in preparation thanks to a team from our Discord and in a room that will be dedicated to you. All your delusions, lies and thefts will be exposed and broadcast on all networks (Facebook, Reddit, GitHub, etc...). You should admit that you took over my code by attributing paternity to yourself and correct your file headers by restoring my rights. Enjoy your pseudo glory for a few more days. After that it will be very complicated for you. NB: since your appearance here, we have taken the time to clone your repositories if you had the idea of ​​making the evidence disappear.

LCh-77, 2022-08-25:
> Confirmed here too. Thank you for fixing this so quickly. I am a beta-tester for an Aviation software company and I have found that making videos to demonstrate a problem helps the developers to see what's wrong.

You are welcome.

tititopher68, 2022-08-26:
@LCh-77 Your so-called fix is ​​bogus...isn't it Doctor Jekyll or Mister Hyde?...in fact ergonomicmike == LCh-77?!!!  You invent a second personality to exist... Schizophrenia!