Jyers/Marlin #2030: if ENABLED(LIMITED_MAX_ACCEL_EDITING) not working?

Ergonomicmike, 2022-08-27:
### Description

LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced as of today.

Ender 3 V2, v4.2.2

Perhaps I'm not understanding correctly how the code is supposed to work. Here's a snippet of the vanilla code in question:

From Configuration.h:

```
/**
 * Default Max Acceleration (speed change with time) (linear=mm/(s^2), rotational=°/(s^2))
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z [, I [, J [, K...]]], E0 [, E1[, E2...]]
 */
#define DEFAULT_MAX_ACCELERATION      { 500, 500, 100, 1000 }  // Ender Configs

//#define LIMITED_MAX_ACCEL_EDITING     // Limit edit via M201 or LCD to DEFAULT_MAX_ACCELERATION * 2
#if ENABLED(LIMITED_MAX_ACCEL_EDITING)
  #define MAX_ACCEL_EDIT_VALUES       { 6000, 6000, 200, 10000 } // ...or, set your own edit limits
#endif
```

If I uncomment 

`#define LIMITED_MAX_ACCCEL_EDITING`

then I expect to be able to set my E-Accel up to 20000 if I want.

But I can only increase E-Accel to 2x the Default value.

### Steps to Reproduce

Uncomment `//#define LIMITED_MAX_ACCCEL_EDITING`

in Configuration.h and compile.

**Expected behavior:** 

I expect to be able to increase my Max E Accel to 20000.

**Actual behavior:** 

I can only increase my Max E Accel to 2000, which is 2x of DEFAULT_MAX_ACCELERATION.

#### Additional Information

So it seems as though the line to define the MAX_ACCEL_EDIT_VALUES is being ignored, and the line before it, which the comment says increases the Defaults to 2x is controlling.

For now, as a workaround, I simple set the Default E-Accel to 10000 and then dial it down after I flash the firmware.

LCh-77, 2022-08-27:
MAX_ACCEL_EDIT_VALUES is not used in JYERSUI.

The editing values are taken from MAX_ACCEL_EDIT_VALUES:

https://github.com/Jyers/Marlin/blob/7b6ff2ce11e614a0c00c472bdfb694fa96d290ab/Marlin/src/lcd/e3v2/jyersui/dwin.cpp#L139-L144

And maximums defined as MAX_ACCEL_EDIT_VALUES x 2:

https://github.com/Jyers/Marlin/blob/7b6ff2ce11e614a0c00c472bdfb694fa96d290ab/Marlin/src/lcd/e3v2/jyersui/dwin.cpp#L2756

Ergonomicmike, 2022-08-27:
Thanks again. As in my previous issue, perhaps a comment in the example Jyers Configuration.h then that Max Accel Edit Values are not used.

closed 2022-08-27 by Ergonomicmike

LCh-77, 2022-08-27:
Marlin configuration files are very generic because they are used by hundreds of printers, you must learn how to use the **Search** function in VSCode to know if a value is used or not in JYersUI.