Jyers/Marlin #2031: #define LIMITED_MAX_FR_EDITING not working?

Ergonomicmike, 2022-08-27:
### Description

LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced code base as of today.

Ender 3 v2 v4.2.2 Configuration.h

As with my previous Issue Report, it appears that LIMITED_MAX_FR_EDITING is not working.

And there might be an inconsistent value setting.

Here's the code:

```
#
define DEFAULT_MAX_FEEDRATE          { 300, 300, 25, 60 }  // Ender Configs

//#define LIMITED_MAX_FR_EDITING        // Limit edit via M203 or LCD to DEFAULT_MAX_FEEDRATE * 2
#if ENABLED(LIMITED_MAX_FR_EDITING)
  #define MAX_FEEDRATE_EDIT_VALUES    { 600, 600, 10, 50 } // ...or, set your own edit limits
```
Notice that the FEEDRATE for Z is lower in the MAX_FEEDRATE value (10) than it is in DEFAULT_MAX_FEEDRATE (25).
This seems inconsistent to me.

### Steps to Reproduce

Uncomment `//#define LIMITED_MAX_FR_EDITING`

and set MAX_FEEDRATE_EDIT_VALUES {600, 600, 25, 200}

in Configuration.h and compile.

**Expected behavior:** 

I expect to be able to set Max E speed to 200.

**Actual behavior:**

I am limited to 120, which is 2x the default value of 60. 

#### Additional Information

The comment for LIMITED_MAX_FR_EDITING says that it sets the new max default to * 2.

So it appears that this line of code is working. But the Max_Feedrate_Edit_Values is not.

LCh-77, 2022-08-27:
MAX_FEEDRATE_EDIT_VALUES is not used in JYERSUI.

The editing values are taken from DEFAULT_MAX_FEEDRATE:

https://github.com/Jyers/Marlin/blob/7b6ff2ce11e614a0c00c472bdfb694fa96d290ab/Marlin/src/lcd/e3v2/jyersui/dwin.cpp#L139-L144

And maximums defined as DEFAULT_MAX_FEEDRATE x 2:

https://github.com/Jyers/Marlin/blob/7b6ff2ce11e614a0c00c472bdfb694fa96d290ab/Marlin/src/lcd/e3v2/jyersui/dwin.cpp#L2696

Ergonomicmike, 2022-08-27:
Okay, thanks. Perhaps a comment in the example configuration.h then to state that Max Feedrate Edit Values are not used in Jyers.

closed 2022-08-27 by Ergonomicmike

LCh-77, 2022-08-27:
> Okay, thanks. Perhaps a comment in the example configuration.h then to state that Max Feedrate Edit Values are not used in Jyers.

Marlin configuration files are very generic because they are used by hundreds of printers, you must learn how to use the **Search** function in VSCode to know if a value is used or not in JYersUI.