Jyers/Marlin #2032: NOZZLE_PARK_POINT math not working?

Ergonomicmike, 2022-08-29:
### Description

LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced
Configuration.h for the Creality Ender 3 v2 v4.2.2 Creatliy UI.

A very small nit, which can be worked around simply by setting the correct values in the UI after the firmware has been flashed:
The NOZZLE_PARK_POINT values are not adding and subtracting correctly.

### Steps to Reproduce

In Configuration.h confirm that

```
#define Y_BED_SIZE 220
#define X_MIN_POS 0
```

(for the Ender 3 v2).

Compile.

**Expected behavior:**

The algebra in the code is

`  #define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 20 }`

Since Y_MAX_POS is defined as Y_BED_SIZE, here 220 then

Nozzle Park in the UI should default to 

   0 + 10 = 10
220 - 10 = 210

to show 10, 210, 20 in the UI. (Z is hard set in the code.)

**Actual behavior:** 

240,220,20

#### Additional Information

I did a search for the term "NOZZLE_PARK_POINT" and only see it one other place in the code, where it is written to EPROM.

tititopher68, 2022-08-30:
@Ergonomicmike (? Or...)
It's very easy to fix. In your jyenhanced.h, delete #define DEF_NOZZLE_PARK_POINT {...} And replace by : constexpr xyz_int_t DEF_NOZZLE_PARK_POINT 
 NOZZLE_PARK_POINT
In dwin.cpp, replace 
eeprom_settings.Park_point = xyz_int_t DEF_NOZZLE_PARK_POINT;
By :
eeprom_settings.Park_point = DEF_NOZZLE_PARK_POINT;

Ergonomicmike, 2022-08-31:
Thank you. For now I simply changed the value in the hardcod in my jyenhanced.h file to correct values for the Ender 3 v2.

LCh-77, 2022-08-31:
> ### Description
> LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced Configuration.h for the Creality Ender 3 v2 v4.2.2 Creatliy UI.
>
> A very small nit, which can be worked around simply by setting the [...]

Great find!, That bug is similar to this:
https://github.com/mriscoc/Ender3V2S1/issues/316
I will correct the code later. Thanks!

tititopher68, 2022-08-31:
@LCh-77 but yes, yes, it can be updated if you make changes in the physical settings....it should be obvious for the modification to be made, you God of coding!  But if you only know how to copy what others are doing...that's for sure, it's much more difficult.

LCh-77, 2022-09-07:
Mmm, you don't know how `constexpr` work...
Jyers is the author of the firmware, others are mere contributors...

[image]