Jyers/Marlin #2034: Blue screen of death on Display update

TMB-HUB, 2022-08-31:
### Description

I have tried to update my ender 3 V2 with Jyres by following a youtube video. The first part (updating the firmware of the mainboard with [E3V2-Default-v4.2.2-v2.0.1.bin](https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-Default-v4.2.2-v2.0.1.bin) was easy. Once I tried to update the display I received something like a blue screen of death on windows. I have followed the instructions carefully and have tried it couple times but I always got the same result.

I did remove the ' (Gotcha') from the folder name.
It was the only folder on the SD card.
I did format my SD to FAT32 with 4k allocatons, but the result is the same.

### Steps to Reproduce

It is always reproducible on my device, it happens all the time.

#### Additional Information

* The video I referred to is [here](https://youtu.be/o1zXYwgIPLY?t=605) 
* I have attached a photo about the Blue screen
* Also attached a photo of the PCB of the display
[image]
[image]

Ergonomicmike, 2022-08-31:
This isn't a problem with the code. Rather it is caused by a revision to the Display board by Creality.

And your display board is unlike any other I've seen. Is this for the Ender 3 v2? 

In any event, it sounds like you're using an (old?) jyers DWIN_SET(Gotcha.)

Instead, you need to find - in the firmware that you're using  - the folder for the LCD called "private." Put that on your SD card and try again. (No DWIN_SET.))

TMB-HUB, 2022-08-31:
> And your display board is unlike any other I've seen. Is this for the Ender 3 v2?

yes, I have received this device (my first printer) earlier this week. It is definitely an ender 3 v2, with mainboard version 4.2.2

> Instead, you need to find - in the firmware that you're using - the folder for the LCD called "private." Put that on your SD card and try again. (No DWIN_SET.))

Could you elaborate on that? The video I've seen referes to the [releases page](https://github.com/Jyers/Marlin/releases). According to the instructions I have downloaded the BIN file and the source code zip. The bin updated the firmware of the mainboard just fine, the display files are in the zip under \Marlin-2.0.1\Display firmware\

I have found [this](https://github.com/Jyers/Marlin/discussions/1768#discussioncomment-2078494) forum and I have tried to follow the comment marked as a solution. (the zip attached to it indeed has a private folder). Still the same result though.

tititopher68, 2022-08-31:
@ @TMB-HUB 
You have a SYNWIT Display ! No chance. Creality provided a few such screens before switching to DACAI. This screen cannot be updated...Sorry. At best pretend a problem with the screen and ask Creality support to replace it.

TMB-HUB, 2022-08-31:
> @TMB-HUB You have a SYNWIT Display ! No chance. Creality provided a few such screens before switching to DACAI. This screen cannot be updated...Sorry. At best pretend a problem with the screen and ask Creality support to replace it.

Hmm... interesting to see that the mfg date of the PCB is this March. Maybe Creality also got hit by display sortages.