Jyers/Marlin #2045: Cant flash new firmware

AkARIO1224, 2022-09-29:
I have an ender 3 v2 and im trying to update my firmware. I searched up how to do it and I believe I did everything properly. Formatted the sd card to FAT32. Checked which version of printer I have (4.2.2). Downloaded the correct files onto the SD card, eject, put in sd card while the printer is off, turn it on. When I turn the printer back on the Creality logo pops up immediately and doesn't properly boot. I left it to run for over 3 minutes and the logo was still there. Does anyone know how to fix this?

rooksoup, 2022-10-09:
I've just encountered the same issue while following the same steps. Made sure to format my SD card to FAT32 with 4k partitions before adding the firmware bin.

astronomz, 2022-11-15:
I have an Ender 3d Max with 4,2.2 motherboard and an ESD4K LCD display.  I installed a CR touch.  Creality firmwares have not worked well with the CR Touch.  Based on favorable comment from a YouTube search, this Jyers firmware seems like the way to go.
Here are some of the things I have tried:
1.  Like "rooksoup" comment on 10/9, I tried all that he attempted. except I was unsuccessful when powering off and then on.
2.  I formatted with FAT32,  Nerver 4k allocation size, because it was not available on two seperate SD's (2k was max on one SD and 8K was min on the other SD.)
3. Do I need a different LCD display?
4. Is flashing of display firmware required before printer firmware?
5. Does this firmware work for a Ender 3d Max?

e87131, 2023-03-29:
@AkARIO1224 
This has worked for me twice already; If you're using firmware with a very long file name, **rename it to something short**.
Using E3V2-BLTouch-3x3-HS-v4.2.2-v2.0.1 did not work for me, but renaming it to 3x3hsbl.bin did.