Jyers/Marlin #2047: JyersUI_Enhanced crashes and reboots at end of Scrolling

Ergonomicmike, 2022-10-03:
### Description

I am running this firmware: github.com/LCh-77/Marlin/tree/bugfix-2.1.x_JYERSUI-Enhanced on an Ender 3 v2. (REC chipset.)

When I go to the Print Menu and I select a short named print file - one that scrolls in the display - the firmware crashes (and then reboots) a few seconds after the end of the scroll.

See this video: https://streamable.com/f4udqo)

### Steps to Reproduce

As per the video, go to the Print Menu and select a short named file that will scroll in the display.

Note that I do NOT have any image bitmaps associated with the print files.

**Expected behavior:** [What you expect to happen]

With old Jyers, nothing happened at the end of the scroll.  (That is, no crash and reboot.) IIRC, the file name repeated scrolling again.

**Actual behavior:** [What actually happens]

It crashes and reboots.

### Additional Information

My display has the rev 1.2 board, which I read does not support the image bitmaps.

Ergonomicmike, 2022-10-07:
Is it just me? Or is anyone else seeing this behavior?

MrAndersson78, 2022-10-26:
I have the exact same issue