Jyers/Marlin #2068: I think this fork is abandoned for good, and good alternative

dhupee, 2022-12-14:
This is also pointed out in open issue #2011 which may be confirmed indirectly after more than a year since the last release, also with the last commit being May this year I can safely say this one is abandoned

if you want to add your custom firmware, the link below might be a good alternative for you

https://github.com/mriscoc/Ender3V2S1

for Jyers thank you for making this, whatever happened to you I just want to wish you the best

perelin, 2023-01-31:
A bit sad. So far was my favourite firmware.

slabua, 2023-02-06:
I was wondering the same

perelin, 2023-05-25:
@RV-from-be this firmware suggested by @dhupee is pretty good already. Running on my EnderV2 for a few month with no issues and regular development https://github.com/mriscoc/Ender3V2S1

RV-from-be, 2023-05-26:
It is sad to always see that some people prefer the easy way without worrying about compliance with licenses and ethics. Mriscoc has always taken malicious pleasure in posing as a victim of so-called copiers when he is the first to do so and to appropriate the code of others. Like recently the one for the development for the laser, code (which for once was provided by Creality) which he appropriated and hid a large part of it in libraries, again violating the GPL license. 3 months ago, I reported, with supporting documentation, the inaccurate statements he made about another developer to the Github team, which validated my request. Mriscoc had to backtrack. In retaliation, he blocked me on his Github. Ask to have access to all of his source code and denounce his disrespect of the GPL license as many have done for 2 years on Reddit (where he hides under the username of a moderator), facebook etc.. .have earned all these people without exception to be banned or blocked. So for me it will always be bad firmware. This developer complains about Creality for its failures towards open source, while he is the first to do like them! The hospital that mocks charity.

dhupee, 2023-05-26:
Well that's tbh is beyond my side, not that i "prefer easy way" or anything, simply that i haven't and don't bother with that for the time being

What i looked up for that alternative is the update and technical performance over other choices

I don't even deal with this person even once, hell i don't even know who you are and your history with this person, i even just know that firmware after i saw a YouTube video

GPL license is a gray area and for now, i personally don't want to deal with the drama, it's your choice to pick it or not, i don't even know if jyers still active or not until i saw your comments

mriscoc, 2023-05-26:
> Mriscoc has always taken malicious pleasure in posing as a victim of so-called copiers when he is the first to do so and to appropriate the code of others.

That is not true, that was a false asseveration from a fake programmer already banned from many sites, he spread that idea using several nick names in many places but never prove that, the only true is that I, in the beginning, protect my code by only releasing it as PR in the official Marlin repository and not posted direct compileable sources, because I detect his "copy & paste" not only from me, also from others firmware programmers.

Some of that history is explained here:  https://github.com/mriscoc/Ender3V2S1/discussions/98

dhupee, 2023-05-26:
to be very honest @mriscoc i love your firmware, your technical side on the firmware is great, its on my printer rn and its working great.

but tbh i cant deal or learn about the drama, sorry if you seems dragged from this okay, but hey...its your problem and you had to deal it, still cool firmware

i will close it to prevent any drama

closed 2023-05-26 by dhupee

RV-from-be, 2023-05-27:
All the art of the "Magician", diverting attention while he does his sleight of hand. So answer this question: how to copy a code that the "Magician" has hidden and pre-encoded (invisible in clear code!)? Either the "fake dev" has supernatural powers, or more simply, he coded the functionalities (admittedly taking a little pleasure in naming functions and variables with the same name, to titillate the ego of the "Magician" as we explained our "fake dev" a long time ago. As for his other assertions (banning on many sites...), the community laughs, it's so stupid and inaccurate. It also has the support of Creality, and others dev in particular those of Klipper and the next build of the "Fake dev" will bring surprises that the "Magician" will try to copy. Good end of the day and end of my right of reply (which the "Magician" does not allow on the sites where he banishes that which is not in accord with him).