Jyers/Marlin #2078: I have installed the E3V2-BLTouch-3x3-v4.2.2-v2.0.1.bin. I have also tried the 5x5 probe area as a shot in the dark. The issue I am having is when running the mesh or homing I do not get Z travel. X and Y perform fine while homing but I then receive a Printer Kill Reason of "Homing Failed, Restart Required". I have tried to manually step the Z with not response. It appears my z is no longer reading. Has anyone ran into this issue and have a solution?

BGood320, 2023-03-07:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Vinced1201, 2023-03-19:
I ran into the same issue on my Ender 3 V2. My Z would only travel up no matter which way I moved the nob.  I will re-flash a new or different version and try it again.

erickrawczyk, 2023-07-21:
@BGood320 Can you at least try and format the issue properly? Long titles like this are really annoying, and failing to fill out the issue template makes this incredibly hard to debug. I understand it's frustrating and you want help quickly, but formatting issues like this dissuades people from helping.

That being said, this sounds like your z-axis stepper motor isn't getting signal. Let's check the basics; make sure all of these are true:
- You have the 4.2.2 board
- you have a bl/cr touch probe, and it's connected to the motherboard and mounted properly
- your z-axis limit switch is disconnected from the motherboard and uninstalled from the printer
- the z stepper motor is connected to the motherboard (I was dumb and disconnected this instead of the z axis limit switch when I installed my probe)

things to test:
- if you re-flash the [stock firmware with touch probe](https://www.creality.com/pages/download-ender-3-v2), does the z motor work again? If it still doesn't work, it's not a firmware issue and the firmware is not the culprit
- If you open up the bottom and expose the motherboard, is your z step motor connected? Is the cable connected at the motor? does using a different cable work?

littlefrank90, 2023-09-16:
Having the SAME EXACT issue, tried installing the original firmware back and works like before.
There is definitely something wrong with the 4.2.2 firmware Z homing.